cesanta/mjs #2: Questions about bindings.

creationix, 2017-01-27:
Just for fun, I decided to see if I can bind to various libuv functions to make a really light-weight node.js.

As a test, I tried one of the simpler functions:

```c
int uv_exepath(char* buffer, size_t* size)
```

At a high level, this takes no arguments and returns a string, but this is C and memory management is hard in C.

In JavaScript I'd ideally have a function like this:

```js
let path = exepath();
```

I wrote something that works, but it leaks memory and is ugly.

```c
static char* exepath() {
  size_t size = 2*PATH_MAX;
  char exe_path[2*PATH_MAX];
  int ret = uv_exepath(exe_path, &size);
  if (ret < 0) return "[error]";
  exe_path[size] = 0;
  char* path = malloc(size + 1);
  for (int i = 0; i < size; i++) {
    path[i] = exe_path[i];
  }
  path[size] = 0;
  return path;
}
```

What are the rules about returning strings from C and how does ownership work?  In all the scripting languages I've ever worked with before, the C code lends a buffer which the VM copies into itself and manages using the GC.  The C code then has ownership of the original again.

I don't see any way to do this here?  Does this language engine simply not care about strings in this manner?

cpq, 2017-01-27:
Now, JS makes it's own copy of the string. If C function returns a malloced char *, it's leaked.
If the returned value is `void *`, you can `free()` it from JS, but then you can't use it as a string.
I guess it is possible to use some ugly workaround:

```
// C
char *tostr(void *ptr) { return (char *) ptr }

// mJS
let exepath = ffi("void *exepath()");
let free = ffi("void free(void *)");
let tostr = ffi("char *tostr(void *)");

let ptr = exepath();
let path = tostr(ptr);  // Makes a JS copy
free(ptr);
```

creationix, 2017-01-28:
Luajit's FFI handles this pretty well with outargs.  This matches how C normally works.  http://luajit.org/ext_ffi_tutorial.html

A really basic way to do this without adding a full C-Types system would be to allow passing in arrays with a single value to outargs.  The first value in JS would be coerced and passed in.  The final value in C would be then coerced and set back into the array.

This particular example is still tricky since we don't have a mutable buffer type (unless you add typed arrays).  A JS array of numbers sounds pretty expensive and strings aren't mutable.

Hmm...

cpq, 2017-03-02:
Now we can create a JS string backed by the `void *`. See `fstr` in the readme.
Also, we allow arbitrary memory access through void pointers.

closed 2017-03-02 by cpq