cesanta/mjs #3: Does mJs support UART on ESP8266/ESP32?

aifer2007, 2017-02-03:
Does mJs support UART on ESP8266/ESP32?

pirj, 2017-03-18:
How does this relate to mjs itself? mjs [provides FFI](https://github.com/cesanta/mjs#cc-interoperability), and the rest is board dependent:

    let f = ffi('int gpio_write(int, int)');  
    f(2, 1);

pirj, 2017-03-18:
You might be interesed in [this example](https://github.com/cesanta/mongoose-os/blob/master/fw/examples/mjs_hello/fs/init.js).

cpq, 2017-03-21:
thanks @pirj

cpq, 2017-03-21:
@aifer2007 not  made api_uart.js yet, you can do it yourself for the time being.

Sources are at https://github.com/cesanta/mongoose-os/blob/master/fw/src/mgos_uart.h
Functions 

```
size_t mgos_uart_write(int uart_no, const void *buf, size_t len);
void mgos_uart_set_write_enabled(int uart_no, bool enabled);
void mgos_uart_flush(int uart_no);

```

Reading from the UART

closed 2017-03-21 by cpq