cesanta/mjs #4: Issue in ffi parser?

creationix, 2017-02-13:
I've made a small toy environment for controlling strips of neopixels.

I think I hit a bug in the FFI parser.

With the following JS:
```js
let fade = ffi("void fade(int keep)");
let update = ffi("void update()");
let rgb = ffi("void rgb(int p, int r, int g, int b)");
let rgbl = ffi("void rgbl(int p, int r, int g, int b, int keep)");
let hue = ffi("void hue(int p, int h)");
let msleep = ffi("void msleep(int ms)");

let i = 0;
let j = 0;
while (true) {
  hue(i, j);
  update();
  i = (i + 1) % 10;
  j = (j + 13) & 768;
  msleep(33);
}
```

It crashes with the error:

```
failed to call FFI signature "void hue(int p, int h)": wrong arg type: failed to parse val type "int p"
```

I don't see any error with this line.  I'm using the same syntax and conventions of the other ffi lines that appear to work.

creationix, 2017-02-13:
Oh I see, it doesn't parse till first call. I guess all my ffi defs must be wrong somehow.

creationix, 2017-02-13:
I looked at the source and realized it doesn't accept the variable names like C code does.  I removed them and just had int and the parsing part worked.

cpq, 2017-03-02:
Yeah, i guess we should fail fast if var names are given.
Or, better, we should accept var names.

closed 2017-03-02 by cpq