cesanta/mjs #6: how export function using c code

fkeujjpdc, 2017-03-02:
i want export function using c code ,because ffi using much rom ,but i'm not find method,like this https://github.com/cesanta/v7/blob/master/examples/call_c_from_js/call_c_from_js.c

cpq, 2017-03-02:
```
 mjs_set(mjs, g, "myfunc", ~0, mjs_mk_foreign(mjs,my_func));
```

See example in https://github.com/cesanta/mjs/blob/master/mjs.c, search for mjs_init_globals

fkeujjpdc, 2017-03-03:
thans for you reply,this is my code ,but result is error:
`#include "mjs.h"

void foo(int x) {
	printf("Hello %d!\n", x);
}

void *my_dlsym(void *handle, const char *name) {
	//if (strcmp(name, "foo") == 0) return foo;
	return NULL;
}

int main(void) {
	struct mjs *mjs = mjs_create();
	mjs_set_ffi_resolver(mjs, my_dlsym); 
	mjs_set(mjs,mjs_get_global(mjs), "foo", ~0, mjs_mk_foreign(mjs, foo));
	//mjs_set(mjs, g, "myfunc", ~0, mjs_mk_foreign(mjs, my_func));
	mjs_exec(mjs, " for(let i=0;i<10;++i) foo(1234);", NULL);
	return 0;
}`
this is result:
Hello 2922240!
Hello 2922240!
Hello 2922240!
Hello 2922240!
Hello 2922240!
Hello 2922240!
Hello 2922240!
Hello 2922240!
Hello 2922240!
Hello 2922240!
请按任意键继续. . .

the function param seems error

cpq, 2017-03-03:
your function needs to take the parameter from the JS stack.
See the example I gave you, how those functions take parameters.

closed 2017-03-03 by cpq