cesanta/mjs #7: mJS FFI not working (ESP8266)

vinogradov, 2017-03-19:
Example from the [documentation](https://github.com/cesanta/mjs/blob/master/README.md):
```javascript
let printf = ffi('int printf(char *, int, char *)');
printf('Hi %d %s\n', 123, 'foo');
```
Result: `MJS callback error: failed to call FFI signature "int printf(char *, int, char *)": cannot resolve function � : failed to call FFI signature "int printf(char *, int, char *)": cannot resolve function m : failed to call FFI signature "int printf(char *, int, char *)": cannot resolve function } : failed to call FFI signature "int printf(char *, int, char *)": cannot resolve function � 
`

cpq, 2017-03-19:
yes, printf example is really bad.

I've just changed it to something more stable and correct, `floor()`.

closed 2017-03-19 by cpq

vinogradov, 2017-03-19:
Thanks for the replay.
Since it's not possible to use `printf` (and it seems `sprintf` too) could you recommend how to convert numbers to strings or concatenate numbers with strings in mJS?

cpq, 2017-03-19:
use JSON.parse/stringify - the "ultimate convertor".

for example, stringify and concatenate two things of any type:

let s = JSON.stringify(my_obj) + JSON.stringify(12345);