cesanta/mjs #8: How can I extend mjs

dodoma, 2017-03-20:
hi,

    There is no problem when use mjs. But How can I modify it? e.g. include var keyword, or String functions.

And I find It's hard to debug mjs(the source code is mixed by multi file).

Any information or suggest about this? Thanks.

cpq, 2017-03-21:
We plan to export unamalgamated sources soon, and it'll be easy then.
Could you elaborate more on what project you're working on, please?

dodoma, 2017-03-21:
Currently, I work on a stock follow project. It's run on Linux x86_64.
I wan't to extend my program by js so others can make application develop using js.

Should be better to use other inline js engine, but I prefer tiny tools, so I can modify it when I need.

mjs Is ok now for my application, If the unamalgamated sources aviable, please let me know, I'm following you on github.

closed 2017-03-21 by dodoma