cesanta/mjs #9: How to iterate over an object?

dodoma, 2017-03-22:
Can't find the js example for interate over array or object, how to do this?

dodoma, 2017-03-22:
```js
function(anArray) {
    for (let i = 0; i < anArray.length; i++) {
        print(anArray[i]);
    }
}
```

Can iterate over an array, But over an object still not find.

cpq, 2017-03-23:
Not possible now - the readme states `No for..in`. 

Is that something critical for you? Could you elaborate on your project please?

dodoma, 2017-03-23:
See it. Not very critical, but may more convinionce for me.

I have two sets of stock orders(may be 40 or more), I need to iterate each one set, and find matches in another set(through id key).
Can be done use two big array, but when thy are object, will be more convenionce.

cpq, 2017-03-30:
`for .. in` will be done, ETA - next week

dodoma, 2017-03-30:
that is great!

cpq, 2017-04-04:
`for..in` implemented.

closed 2017-04-04 by cpq