cesanta/mjs #13: Arrays: type problem.

0rel, 2017-03-30:
Hi, mJS looks very promising, running it on quite restricted micro (STM32L0)!
But I have some problems with using arrays.

```
let a=[2,3,4];
print(a[1]);
```

The elements seem to have a strange type... is it float? It can't access them properly (some times stalls my micro, need to investigate)...

Are there any examples for the use of arrays with mJS already? (I have a hard time understanding the code at the moment... - Do plan to release it also as "editable" Project (with the source files separated out?)

cpq, 2017-03-30:
Yes, we plan to open non-amalgamated files fairly soon - today/tomorrow.

Together with that, we'll release the refactoring of the engine. Currently it uses an internal forth-like machine. The refactoring will use direct compilation from the parser to bcode. It's likely that your problem will go away, so I suggest to wait until tomorrow.

0rel, 2017-03-30:
That's great to hear! I'm looking forward to try out the new version, and give you feedback. Concerning amalgamation: I was already looking into mjs2 for a bit, it's definitely nicer to read/edit the files in a split form.

One minor problem I had while testing mjs on the hardware (STM32L072, 192 KiB Flash, 20 KiB RAM), was that that bare printf()/fprintf()/ putc() calls are in the sources. Those had to be replaced manually, otherwise it runs quite nicely already (interactive REPL-style), quite happy about that! :)

cpq, 2017-03-30:
Thanks !
What's your suggestion w.r.t. those printf calls ?

0rel, 2017-03-30:
Since mjs is really about keeping it tiny and simple, I don't know if this is the right thing to do, but [https://github.com/zsaleeba/picoc](picoc) which I discovered today, tries to solve it with platform specific files, the user has to provide/modify, like there: https://github.com/zsaleeba/picoc/tree/master/platform. That might be a very explicit solution but adds too much clutter...

Maybe it's better to use simple LOG_*() macros to replace the calls? ... Another problem I had with the stdout/stderr was that I had to replace some newline characters \n  here and there by \r\n, because my UART console somehow requires to send both... But that's a detail. - It's really cool that the error messages point out where the syntax error happens, added a lot to the interactivity of the mjs console...

ATM there are some ugly fixes in there like that to get the logs working (mostly only an issue on a embedded platform):
`
extern void console_log_fmt_LOG_TMP(const char *format, ...); // tmp
extern void console_log_fmt_LOG_TMP_NEWLINE(); // tmp

// ...

void bf_print_cell(struct bf_vm *vm, bf_cell_t cell) {
  struct mjs *mjs = (struct mjs *) vm->user_data;
  mjs_val_t v = (mjs_val_t) cell;
  if (mjs_is_number(v)) {
    console_log_fmt_LOG_TMP("%d", mjs_get_int(mjs, v));
  } else if (mjs_is_string(v)) {
    size_t size;
    const char *s = mjs_get_string(mjs, &v, &size);
    console_log_fmt_LOG_TMP("%.*s", (int) size, s);
  } else if (mjs_is_undefined(v)) {
    console_log_fmt_LOG_TMP("undefined");
  } else if (mjs_is_null(v)) {
    console_log_fmt_LOG_TMP("null");
  } else if (mjs_is_boolean(v)) {
    if (mjs_get_bool(mjs, v)) {
      console_log_fmt_LOG_TMP("true");
    } else {
      console_log_fmt_LOG_TMP("false");
    }
  } else if (mjs_is_object(v)) {
    console_log_fmt_LOG_TMP("[object Object]");
  } else {
    console_log_fmt_LOG_TMP("<val_t %llx>", (long long unsigned int) v);
  }
  /*
  struct mjs *mjs = (struct mjs *) vm->user_data;
  mjs_val_t v = (mjs_val_t) cell;
  if (mjs_is_number(v)) {
    printf("%d", mjs_get_int(mjs, v));
  } else if (mjs_is_string(v)) {
    size_t size;
    const char *s = mjs_get_string(mjs, &v, &size);
    printf("%.*s", (int) size, s);
  } else if (mjs_is_undefined(v)) {
    printf("undefined");
  } else if (mjs_is_null(v)) {
    printf("null");
  } else if (mjs_is_boolean(v)) {
    if (mjs_get_bool(mjs, v)) {
      printf("true");
    } else {
      printf("false");
    }
  } else if (mjs_is_object(v)) {
    printf("[object Object]");
  } else {
    printf("<val_t %llx>", (long long unsigned int) v);
  }
  */
}`

cpq, 2017-04-04:
non-amalgamated sources, and tests, are published now.

w.r.t. printf calls - what is the problem of routing that to uart? we do exactly that in github.com/cesanta/mongoose-os

closed 2017-04-04 by cpq

0rel, 2017-04-07:
Thanks! Will be testing it soon...