cesanta/mjs #14: invalid JSON string on JSON.parse() when object has more than 20 nodes

dodoma, 2017-03-31:
As mentioned on title, tested under osx and x86_64 linux.

Here is my exe.c

```c
#include <dlfcn.h>

#include "mjs.h"

int main(int argc, char *argv[])
{
    if (argc < 2) {
        printf("useage: %s js file\n", argv[0]);
        return 1;
    }

    struct mjs *jsm = mjs_create();
    mjs_set_ffi_resolver(jsm, dlsym);

    mjs_err_t jserr;
    mjs_val_t res;

    jserr = mjs_exec_file(jsm, argv[1], &res);
    if (jserr != MJS_OK) {
        printf("%s\n", mjs_strerror(jsm, jserr));
    }

    mjs_println(jsm, res);

    return 0;
}
```

And there are two js file: 1.js, 2.js
```js
let x = '{"a":1,"b":1,"c":1,"d":1,"e":1,"f":1,"g":1,"h":1,"i":1,"j":1,"k":1,"l":1,"m":1,"n":1,"o":1,"p":1,"q":1,"r":1,"s":1,"t":1}';

let xo = JSON.parse(x);
```

```js
let x = '{"a":1,"b":1,"c":1,"d":1,"e":1,"f":1,"g":1,"h":1,"i":1,"j":1,"k":1,"l":1,"m":1,"n":1,"o":1,"p":1,"q":1,"r":1,"s":1,"t":1, "u": 1}';

let xo = JSON.parse(x);
```

Here is the output:
    maliang@bogon ~/tmp/x/mjs$ ./exec 1.js
    [object Object]
    maliang@bogon ~/tmp/x/mjs$ ./exec 2.js
    invalid JSON string
    undefined

When I use cc mjs.c -DMJS_MAIN -D_DARWIN_C_SOURCE -o mjsexec, and the later js work ok when it is the command parameter
   maliang@bogon ~/tmp/x/mjs$ ./mjsexec -e "let x = '{"a":1,"b":1,"c":1,"d":1,"e":1,"f":1,"g":1,"h":1,"i":1,"j":1,"k":1,"l":1,"m":1,"n":1,"o":1,"p":1,"q":1,"r":1,"s":1,"t":1, "u":1,"v":1}'; let xo = JSON.parse(x)"
[object Object]

cpq, 2017-04-04:
We've changed mjs core recently, please try again

dodoma, 2017-04-05:
New code don't have this problem.

closed 2017-04-05 by dodoma