cesanta/mjs #16: Can I extend mjs through mjs_op_getprop() ?

dodoma, 2017-03-31:
Say I need to pop out an item from an array, Since there is no prototypes for Array, I couldn't use array.prototype.pop() to do this. 

I notice that there is a function mjs_op_getprop() nearly to it. So, Can I write my own `pop` properity for array val?  This function will push an val to &vm->dstack, but Array.pop() seems don't need(It just modify Array). I couldn't make sure is it possible?

BTW, array[array.length] = newitem works equal to array.push(newitem).

cpq, 2017-04-04:
could you elaborate more on what are you trying to do on a higher level, please?

dodoma, 2017-04-05:
I want to append an item into an array, and, pop it out later. 
This array contains some sorted, dynamic stock price information.

cpq, 2017-04-05:
Isn't this sufficient?

https://github.com/cesanta/mjs/blob/2565635a6bef5a9d884373e2b79186ec62f2cc35/mjs.h#L244-L264

dodoma, 2017-04-06:
Sorry for fix my last reply as following:

    I want to INSERT item into an array(at ARBITRARY position), and, pop it (or them) out later.
    This array contains some sorted, dynamic stock price information.

I tried the mjs.h offered `array related functions`, There should be two more functions to satisfy my need:

```c
mjs_array_insert(struct mjs *mjs, mjs_val_t arr, unsigned long index, mjs_val_t v);
mjs_array_del/remove(struct mjs *mjs, mjs_val_t arr, unsigned long index, unsigned long count);
```

With these functions and FFI, everything related to `array` will be ok.

cpq, 2017-04-06:
```
let a = [1,2,3];
a[1] = 7;
JSON.stringify(a);'
```

You can insert into list. In JS, or in C via `mjs_set`

dodoma, 2017-04-06:
maliang@bogon ~/code/mjs/mjs$ ./build/mjs -e 'let a = [1,2,3]; a[1] = 8; JSON.stringify(a)'
    [1,8,3]

I need result to be [1, 8,2,3].

`a[1] = 8;` is a array item replacement operation, not insert.
and, I need some way to delete items from array.

cpq, 2017-04-06:
ok, you want splice, basically.

dodoma, 2017-04-06:
en, Yes :)

dimonomid, 2017-04-10:
`Array.prototype.splice()` is implemented in https://github.com/cesanta/mjs/commit/a15a10a5716228eb427da12bdc85843f268287d8 , closing this issue. Thanks for the feedback!

closed 2017-04-10 by dimonomid