cesanta/mjs #20: More readable error trace

dodoma, 2017-04-06:
Is there any function to trace the js execution error, like v7_printf_error() in v7? 

I use mjs_str_error() currently, and it's output isn't readable as much as v7_printf_error(), especially error occured with in multipile js files.

Here is my test js files:

1.js
```js
let x = {
    floor: ffi('double floor(double)'),
    floor2: ffi('double floor_unexist(double)')
};
```

2.js
```js
load("tests/1.js")

let y = {
    foo: function(n) {
        return x.floor(n);
    },
    foo2: function(n) {
        return x.floor2(n);
    }
};

y.foo(12.3);
y.foo2(12.3);
```

Here is the error output:
    maliang@bogon ~/code/mjs/mjs$ ./build/mjs tests/2.js 
      at tests/2.js:8
      at tests/2.js:14
    undefined

There are two main problem:
1. `tests/1.js` not appeared in error string.
2. js function names  not appeared in error string.
3. no ffi failed (change mjs_strerror(mjs, mjs->error) to mjs_strerror(mjs, res) will have `failed to call FFI...`)

BTW, If I have some spell mistake in 1.js. e.g.:
```js
let x = {
    floor: ffi('double floor(double)');
    floor2: ffi('double floor_unexist(double)')
};
```
Error trace message can't totoaly understood.
    maliang@bogon ~/code/mjs/mjs$ ./build/mjs tests/2.js 
      at tests/2.js:1
    undefined
If i use mjs_strerror(mjs, res) instead of mjs_strerror(mjs, mjs->error):
  at tests/2.js:1
NO_ERROR
undefined

cpq, 2017-04-11:
That has been improved, please test

closed 2017-04-11 by cpq

dodoma, 2017-04-12:
It's ok now, thanks a lot.