cesanta/mjs #21: i compile mjs by arm-none-eabi-gcc,the hex is very large

fkeujjpdc, 2017-04-10:
i put mjs.h and mjs.c into a empty project ,define MARCO  `#define CS_PLATFORM CS_P_STM32` in mjs.h.
then compile the project by arm-noe-eabi-gcc,the code is very large.

text                  data            bss                dec
110796             2772          356             112924

cpq, 2017-04-11:
That's the size of the whole firmware?
I don't see mjs.o size if your breakdown.

closed 2017-04-11 by cpq

fkeujjpdc, 2017-04-12:
i use the -Os flags:
this is my mjs.o size:
D:\workspace\test\Release\mjs>arm-none-eabi-size --format=berkeley "mjs.o"
   text    data     bss     dec     hex filename
  41928     136       5   42069    a455 mjs.o

the other module size is very small:
   text    data     bss     dec     hex filename
    650       0       0     650     28a stm32f10x_gpio.o
    836      20       0     856     358 stm32f10x_rcc.o
   3202       0       0    3202     c82 stm32f10x_tim.o
    924       0       0     924     39c stm32f10x_usart.o
arm-none-eabi-size: subdir.mk: File format not recognized