cesanta/mjs #23: Running unit tets on Centos 6 32-bit

geert56, 2017-04-11:
I had some trouble running the unit tests to all PASS.
This is on a Centos 6 32-bit machine. I discovered that it is best to use -rdynamic as extra
CFLAGS in the Makefile. Also, I extended the typedefs in ffi.c to include the 2 and 3 argument cases
and added explicit code to detect and handle them. The problem I was seeing that f(int, double)
was not handled correctly and gave me wrong argument values.
Surprisingly, on a Centos x86_64 none of these problems occurred.

cpq, 2017-04-11:
@dimonomid dima could you add a unit test for the int,double case please?

dimonomid, 2017-04-11:
Sure, will do.

geert56, 2017-04-11:
test_func2() was failing for me, among others. That's the test for a call to a function with signature
int f(int, double).
Without -rdynamic, dlsym would not find certain symbols, e.g. mjs_mem_get_uint.

dimonomid, 2017-04-18:
Indeed there was an issue with ffi on 32bit architectures, fixed by https://github.com/cesanta/mjs/commit/072d8550f9d62fa2228727cb7e69dd307caafc60 .

That was an oversight not to run tests on 32bit machines, of course. Thanks for reporting!

closed 2017-04-18 by dimonomid