cesanta/mjs #25: High cpu load on memory increase

dodoma, 2017-04-12:
CPU load almost 100% on mjs manage about 3M memory.

Here is my test code and data, tested on osx x86_64.

load.c
```c
#include "mjs.h"
#include <dlfcn.h>
#include <string.h>
#include <sys/time.h>

int main(int argc, char *argv[])
{
    /*
     * prepare big object
     */
    char strbuf[10000];
    int len = 0;
    FILE *fp = fopen("./x.json", "r");
    if (fp) {
        memset(strbuf, 0x0, sizeof(strbuf));
        len = fread(strbuf, 1, 10000, fp);
        fclose(fp);
        printf("read %d bytes from x.json\n", len);
    } else {
        printf("can't open x.json\n");
        return 1;
    }

    struct mjs *jsm = mjs_create();

    mjs_set_ffi_resolver(jsm, dlsym);

    mjs_err_t jerr;
    mjs_val_t res;
    jerr = mjs_exec_file(jsm, "./x.js", &res);
    if (jerr != MJS_OK) {
        printf("%s\n", mjs_strerror(jsm, jerr));
    }

    mjs_val_t global = mjs_get_global(jsm);
    mjs_val_t object = mjs_get(jsm, global, "myObject", ~0);
    mjs_val_t function = mjs_get(jsm, object, "test_func", ~0);

    struct timeval total_tvs, total_tve;
    gettimeofday(&total_tvs, NULL);

    for (int i = 0; i < 100; i++) {
        struct timeval tvs, tve;
        gettimeofday(&tvs, NULL);

        mjs_val_t a1 = mjs_mk_string(jsm, strbuf, len, 0);

        jerr = mjs_call(jsm, &res, function, object, 1, a1);
        if (jerr != MJS_OK) {
            printf("%s", mjs_strerror(jsm, jerr));
        }

        gettimeofday(&tve, NULL);

        //unsigned long ela = (tve.tv_sec - tvs.tv_sec) * 1000000ul + (tve.tv_usec - tvs.tv_usec);
        printf("call %d elapsed %d seconds\n", i, tve.tv_sec - tvs.tv_sec);
    }

    gettimeofday(&total_tve, NULL);

    printf("totoal elapsed %d seconds\n", total_tve.tv_sec - total_tvs.tv_sec);

    mjs_destroy(jsm);

[... 207 lines cut ...]
call 90 elapsed 4 seconds
92 
call 91 elapsed 6 seconds
93 
call 92 elapsed 14 seconds
94 
call 93 elapsed 17 seconds
95 
call 94 elapsed 19 seconds
96 
call 95 elapsed 18 seconds
97 
call 96 elapsed 20 seconds
98 
call 97 elapsed 25 seconds
99 
call 98 elapsed 20 seconds
100 
call 99 elapsed 20 seconds
totoal elapsed 387 seconds
```

My load:
    PID   COMMAND      %CPU TIME     #TH  #WQ  #POR MEM    PURG CMPR PGRP PPID STATE   BOOSTS     %CPU_ME %CPU_OTHRS UID  FAULT COW  MSGS MSGR SYSB SYSM CSW    PAGE IDLE POWE
    4750  load         98.8 01:27.67 1/1  0    10   2240K  0B   0B   4750 978  running *0[1]      0.00000 0.00000    501  1310  126  36   15   110  63   72816+ 0    0    98.8

cpq, 2017-04-12:
@dimonomid could you take a look please? Maybe GC is interfering. The execution time seems extremely uneven, and indeed shows a big bump starting from certain point.

dodoma, 2017-04-13:
It's gc to eat the cpu load.

I add one line at the bottom of mjs_create()
```c
  mjs->inhibit_gc = 1;
```
these codes will be done in 1 second.

dodoma, 2017-04-13:
Suggest do gc periodly(by time or memory size), If it's hard to short the gc check time.

dimonomid, 2017-04-13:
TL;DR:

Try to do the following: find this line:
```c
#define _MJS_STRING_BUF_RESERVE 100
```
and change 100 to, say, 10000.

And then compile with these flags:
```
-DMJS_DEFAULT_OBJECT_ARENA_SIZE=500 -DMJS_DEFAULT_PROPERTY_ARENA_SIZE=500
```
-------
GC is invoked when preallocated arenas for objects and properties (and buffer for strings) are about to be exhausted.

By default the arenas for object/properties contain 20 objects, and when it's not enough (and GC can't release anything), arena size is incremented by 10.

When strings buffer is exhausted, its size is incremented by 100.

These defaults are probably fine for the embedded environment (which is our primary target), but not so fine for the machine with lots of memory available.

Currently, only initial size of object/property arenas can be tweaked (with the compile flags I mentioned above), other params are hardcoded. We'll need to fix this of course, but for now, could you please perform the modifications I mentioned, and see if it helps?

dodoma, 2017-04-13:
Sounds good, I'll try it tomorrow. And maybe some other questions :)

dodoma, 2017-04-14:
dimonomid, I tried you advice just now(thanks for the detail explain), I suggest to do following modify on mjs gc:

```c
 MJS_PRIVATE int gc_strings_is_gc_needed(struct mjs *mjs) {
   struct mbuf *m = &mjs->owned_strings;
-  return (double) m->len / (double) m->size > 0.9;
+  if ((double)(m->size - m->len) / (double)_MJS_STRING_BUF_RESERVE < 0.1) {
+      mbuf_resize(m, m->size + _MJS_STRING_BUF_RESERVE);
+      return 1;
+  }
+  return 0;
 }
```

It can avoid unnecessary mjs_gc() on:
1. mjs->owned_strings.len / mjs->owned_strings.size `always` greater than 0.9, if mjs->owned_strings.len grows very big after mjs long time running.

2.  If we mark mjs->need_gc = 1; after gc_strings_is_gc_needed(), we'd better grow up owned_string.size right now, to avoid repeat mjs_gc() call if gc_compact_strings() can't do very usefull compact.

dodoma, 2017-04-14:
By the way, I seted both _MJS_STRING_BUF_RESERVE, MJS_DEFAULT_OBJECT_ARENA_SIZE, and MJS_DEFAULT_PROPERTY_ARENA_SIZE to 1048576. And seted arena grow up size to 524288.

This runs fine on my 4GB environment, maybe it will be usefull If we offer some `non-embed` environment documents.

cpq, 2017-04-18:
Can we close this?

dodoma, 2017-04-18:
@dimonomid do what ever you want, it's ok now for me.

dimonomid, 2017-04-18:
Before closing this, I want to make these parameters easily adjustable with the compile flags (without the need to edit the code). Also, this is a relevant point:

> mjs->owned_strings.len / mjs->owned_strings.size always greater than 0.9, if mjs->owned_strings.len grows very big after mjs long time running.

Not sure we'll address it now, but let me think more.

dodoma, 2017-04-18:
En, let's consider the extrimily situation. we push the 'x.josn' into an array for 10000 times. we will reproduce the problem.

cpq, 2017-04-18:
@dodoma mJS focuses on embedded environment. Supporting those corner cases is not our priority.

dimonomid, 2017-04-18:
These memory parameters are now adjustable with compiler flags: https://github.com/cesanta/mjs/commit/3ccc378af6177ace11c8a2b94b129b4127d807e0

Closing this one.

closed 2017-04-18 by dimonomid