cesanta/mjs #26: wrong bcode offset on function parameter TYPE_ERROR

dodoma, 2017-04-13:
Here is the test case:
```
./build/mjs -e "let y = {foo: function(s){return s}, foo2: function(){y.foo('xxx' + 10)}}; y.foo2();"
  at <stdin>:1
  at <stdin>:1
ERROR: wrong bcode offset -2147483648
  at <unknown-filename>:1
Error: TYPE_ERROR
```

This is the js file in pretty format:
```js
let y = {
    foo: function(s) {
        return s;
    },
    foo2: function() {
        return y.foo('xxx' + 10);
    }
};

y.foo2();
```

dodoma, 2017-04-13:
BTW, the stack trace is printed to `stderr` on js->error != MJS_OK, Could I appoint to another FILE* to trace ?

dimonomid, 2017-04-13:
The actual problem is in this line: 
```javascript
return y.foo('xxx' + 10);
```
mjs does not perform autoconversion from a number to a string for you. We're not sure yet if we want it to be implemented. The only way to make it work at the moment is as follows:
```javascript
return y.foo('xxx' + JSON.stringify(10));
```
The "wrong bcode offset" is unrelated to this: it's a problem of printing backtrace. It surely needs to be fixed.

dodoma, 2017-04-13:
ok, expect the backtrace printing fix.

dimonomid, 2017-04-13:
We improved the error message when one does `'xxx' + 10`: instead of plain "type error", the error being returned is "implicit type conversion is prohibited".

Backtrace printing is [fixed](https://github.com/cesanta/mjs/commit/a0e743f9b40a64905cc15dbe830b05d015000bc1) as well. Closing this, and thanks for reporting!

closed 2017-04-13 by dimonomid