cesanta/mjs #28: ffi_exports.c errors

aifer2007, 2017-04-26:
Today, when I build my mos project, I get much more ffi_exports.c erros, like this:

/app/build/gen/ffi_exports.c:146:6: error: conflicting types for 'mgos_conf_value_type'
 int  mgos_conf_value_type(void *);
      ^
/app/build/gen/ffi_exports.c:145:6: note: previous declaration of 'mgos_conf_value_type' was here
 void mgos_conf_value_type(void);

The mos project can be built some days ago.

aifer2007, 2017-04-26:
I have the ffi_symbols: in my mos.yml file.
All the ffi function in ffi_symbols list has a ffi_exports.c erros.
ffi_symbols is outdate?

aifer2007, 2017-04-26:
yes, ffi_symbols is the reason.

dimonomid, 2017-05-23:
ffi_exports used to contain dummy signatures like `void symbol_name(void)`, but now it contains signatures as they are given to the `ffi("..")` function (slightly preprocessed), so I think this issue should be resolved now. Feel free to reopen if it's not. Thanks for reporting!

closed 2017-05-23 by dimonomid

christianmue1980a, 2018-03-14:
I think I observed a very similar issue and described it here. Do you have an idea? https://github.com/cesanta/mongoose-os/issues/414