cesanta/mjs #30: Data lost on call an unexist method

dodoma, 2017-04-28:
say this code:

```c
#include "mjs.h"
#include <dlfcn.h>
#include <limits.h>

int main(int argc, char *argv[])
{
    struct mjs *jsm = mjs_create();

    mjs_set_ffi_resolver(jsm, dlsym);

    mjs_err_t jerr;
    mjs_val_t res;

    jerr = mjs_exec_file(jsm, argv[1], &res);
    if (jerr != MJS_OK) {
        printf("js file error %s\n", mjs_strerror(jsm, jerr));
        return 1;
    }

    for (int i = 0 ; i < 3; i++) {
        mjs_val_t obj = mjs_get(jsm, mjs_get_global(jsm), "obj", ~0);
        mjs_val_t f1 = mjs_get(jsm, obj, "f1", ~0);

        jerr = mjs_call(jsm, &res, f1, obj, 1, mjs_mk_number(jsm, 212));
        if (jerr != MJS_OK) {
            printf("%s\n", mjs_strerror(jsm, jerr));
        }

        if (i == 1) {
            obj = mjs_get(jsm, mjs_get_global(jsm), "obj", ~0);
            f1 = mjs_get(jsm, obj, "funexist", ~0);
            jerr = mjs_call(jsm, &res, f1, obj, 1, mjs_mk_number(jsm, 212));
            if (jerr != MJS_OK) {
                printf("%s\n", mjs_strerror(jsm, jerr));
            }
        }
    }

    return 0;
}
```

With the follow x.js
```js
let obj = {
    arr: []

    f1: function(a) {
        this.arr[this.arr.length] = a;

        //print('first ' + a);
        print(JSON.stringify(this.arr));
    }
};
```

Output:

```
maliang@bogon ~/code/mjs/mjs/tests$ gcc -std=c99 -o x x.c -I ../../ -L ../build/ -lmjs
maliang@bogon ~/code/mjs/mjs/tests$ ./x x.js
[212] 
[212,212] 
[212] 
```

There are 3 problems:

1. parse x.js  `arr: []` whithout `,` terminate don't return error.

2.  call an unexist method `funexist` don't return error.

3.  After `funexist` called, obj.arr[] rested to empty.

cpq, 2017-07-17:
Feel free to send a PR :)

dodoma, 2017-08-07:
Ok, don't know how exactly the data losted.
I add an pre judgement in mjs_appy()

dimonomid, 2017-12-02:
Thanks, fixed in private repo, will be published here soon.

closed 2017-12-02 by dimonomid