cesanta/mjs #32: build

lemonhall, 2017-05-01:


lemonhall, 2017-05-01:
missing some file?? common.mk??

rwhitworth, 2017-05-10:
Also seeing this on Debian.

rwhitworth, 2017-05-10:
Header files are also missing.  This project isn't buildable in its current state.

dodoma, 2017-05-10:
Header files missing is imported by commit 
```
308c23a7 (Dmitry Frank 2017-05-04 17:55:52 +0100  9) #include "common/mg_str.h"
```
Seems it should be fixed soon. Currently you can check out to previous commit
```
git checkout 8fe46f1
```

This works fine.

dodoma, 2017-05-10:
BTW, @lemonhall , 
```
echo "CLANG = gcc" > common.mk
```
on the mjs/ top directory. will  build success.

rwhitworth, 2017-05-10:
@dodoma Thank you for the suggestions!  They produce a working build.

dimonomid, 2017-05-22:
Thanks for reporting that, fixed.

closed 2017-05-22 by dimonomid