cesanta/mjs #33: Segmentation Faults 2017-05-10

rwhitworth, 2017-05-10:
Hello,
I was using American Fuzzy Lop (afl-fuzz) to fuzz input the mjs program on Linux. Is fixing the crashes from these input files something you're interested in? The input files can be found here: https://github.com/rwhitworth/mjs-fuzz/tree/master/2017-05-10.  I used mjs compiled from git commit 8fe46f11c85938814897a5b46a15d137615b1292, as mentioned in issue #32.  If there is a better way to compile this for Linux on x86_64 please let me know so I can use the latest version.

The files can be executed as `./mjs id_filename` to cause seg faults.

Let me know if I can provide any more information to help narrow down this issue.

dimonomid, 2017-05-22:
Thanks for reporting this, most of the test cases you provided are fixed by a few last commits, working on the rest.

dimonomid, 2017-05-23:
Ok, so this is probably the last fix for now https://github.com/cesanta/mjs/commit/3603a08b9867dda63d2f821f0df4878407847db5

We can't make mjs totally safe, because providing pointer operations inherently means adding an ability to shoot oneself in the foot. The rest of the issues are about writing to or reading from invalid addresses, so I don't think we can fix them.

Thanks again for reporting, this uncovered a few nasty corner cases, which are fixed now. Closing.

closed 2017-05-23 by dimonomid