cesanta/mjs #34: Add API for SPI

applecrusher, 2017-05-11:
Now that SPI support has been added to Mongoose OS. Would it be possible to add the API to mjs. Thanks!

sbeattie, 2017-07-24:
+1 Would love to try this out in Javascript too.

dimonomid, 2017-07-24:
ETA: end of this week.

dimonomid, 2017-07-28:
Done, see the JavaScript API reference here: https://mongoose-os.com/docs/libraries/hardware/spi.html#load

Available in latest version only so far, will be integrated in the upcoming release 1.12.

If you're on release (which is a good thing), you can either switch to latest version temporary (`mos update latest`), or just use latest versions of libs `spi` and `mjs` in your `mos.yml`, like that:

```
  - origin: https://github.com/mongoose-os-libs/spi
    version: latest
  - origin: https://github.com/mongoose-os-libs/mjs
    version: latest
```

closed 2017-07-28 by dimonomid