cesanta/mjs #35: ffi exports are not parsed correctly

karimsa, 2017-05-18:
It seems that at build time, when uses of `ffi()` are searched for in the JS code, the search also checks comments.

For instance, the following code can cause the build to error out:

```javascript
// just to kill things: ffi('void non-existent-function(double)')
```

Which means that when debugging, if I want to comment out the `ffi()` use, I have to change the name of the function to `ffi_` or anything else - I can't just comment it out.

Perhaps the defining of `ffi_exports` can be moved into a header file (i.e. `main.h`) or through `main.c` since they are converted to C anyways during the build process. Having the build process pull it out of the JS can lead to other issues as well and gives the illusion of dynamic interoperability - which isn't really the case anyways.

closed 2017-05-21 by dimonomid

karimsa, 2017-05-21:
Why was this closed? Why is there no response at least if you're not interested in fixing this bug?

dimonomid, 2017-05-21:
Thanks for reporting!

The fix is committed to our internal repository, and will be published to https://github.com/cesanta/mongoose-os soon (technically, the issue is about mongoose-os, not the mjs itself). Now, comments are stripped from JS file before parsing ffi signatures.

Sorry for the confusion about closing it before publishing, github does that automatically because commit description mentions this issue.