cesanta/mjs #47: support for mJS with stm32f100R8

johnnycubides, 2017-10-15:
Hi, I'll like help about this;
I like the idea of a interpreter for stm32f100 with 128k of memory ROM.  
I tried compiled for this, and I have this problems:

../mJS/mjs.c:1526:20: fatal error: **memory.h**: No such file or directory
compilation terminated.
Makefile:123: recipe for target 'build/mjs.o' failed
make: *** [build/mjs.o] Error 1

and this is similary for 
**#include <stm32_sdk_hal.h>**

cpq, 2017-10-15:
what sdk includes are available?

johnnycubides, 2017-10-16:
Hi, I generated the code with cubeMX, and these are the files for hal:

Legacy/                  stm32f1xx_hal_cortex.h  stm32f1xx_hal_dma.h       stm32f1xx_hal_gpio_ex.h  stm32f1xx_hal_i2c.h     stm32f1xx_hal_rcc.h     stm32f1xx_hal_tim.h
stm32f1xx_hal_adc_ex.h  stm32f1xx_hal_def.h     stm32f1xx_hal_flash_ex.h  stm32f1xx_hal_gpio.h     stm32f1xx_hal_pwr.h     stm32f1xx_hal_spi.h     stm32f1xx_hal_uart.h
stm32f1xx_hal_adc.h     stm32f1xx_hal_dma_ex.h  stm32f1xx_hal_flash.h     stm32f1xx_hal.h          stm32f1xx_hal_rcc_ex.h  stm32f1xx_hal_tim_ex.h

so, I added the next change in common/platforms/platform_stm32.h
`-#include <stm32_sdk_hal.h>`
`+#include <stm32f1xx_hal.h>`

and in mjs.c
`-#include <stm32_sdk_hal.h>`
`+#include "stm32f1xx_hal.h"
`
and I try it compile again, finally this is the new error 
`In file included from /opt/gcc-arm-none-eabi-6-2017-q2-update/arm-none-eabi/include/dirent.h:7:0,
                 from ../mJS/mjs.c:2201:
/opt/gcc-arm-none-eabi-6-2017-q2-update/arm-none-eabi/include/sys/dirent.h:10:2: error: #error "<dirent.h> not supported"
 #error "<dirent.h> not supported"`.

please, do you have any suggest ?
regards

cpq, 2017-10-18:
Try to compile with
`-DMG_ENABLE_FILESYSTEM=0`

cpq, 2017-10-26:
See https://github.com/cesanta/mjs/issues/51

closed 2017-10-26 by cpq