cesanta/mjs #48: include in esp32 firmware

garfy0886, 2017-10-17:
hello, i would like to integrate mjs into my firmware but i can not understand how to do it. I tried to include the sources but I can not compile. thank you

cpq, 2017-10-26:
Here is an example https://github.com/mongoose-os-libs/mjs

What project you're working on - is it personal or commercial?

garfy0886, 2017-10-26:
for now this part is being tested but if everything works it should become commercial .. why?

cpq, 2017-10-30:
For commercial clients, we provide commercial support, so if you're interested, https://mongoose-os.com/contact.html

Did you manage to build mJS for your ESP32?
Did you consider https://mongoose-os.com/ where mJS is already integrated ?

closed 2017-11-26 by cpq

garfy0886, 2018-01-10:
Sorry for the delay for the reply, I have not yet managed to build mJS for my ESP32. I don't understand how to include the library in my project. can you help me?