cesanta/mjs #50: can i build it vs 2017?

0x410c, 2017-10-24:
i am getting errors building it on visual studio 2017?

has anyone compiled it on windows?

cpq, 2017-10-24:
I think some minor cleanup should be done in order to make it compile on windows.
Specifically, C99 constructs, maybe others.

0x410c, 2017-10-24:
i am able to compile it now, some changes here and there in headers,

but im getting read access violation, after returning from winapi's messageboxa function

code:
```
#include<Windows.h>
#include "mjs.h"
#include <conio.h>

void *my_dlsym(void *handle, const char *name) {
	if (strcmp(name, "msg") == 0) return GetProcAddress(LoadLibraryA("user32.dll"),"MessageBoxA");
	return NULL;
}

int WINAPI WinMain(HINSTANCE hInst,HINSTANCE hPrev,LPSTR cmd,int n)
{
	struct mjs *mjs = mjs_create();
	mjs_set_ffi_resolver(mjs, my_dlsym);
	mjs_exec(mjs, "let f = ffi('int msg(int,char*,char*,int)'); f(0,'asdf','asdf',0);", NULL);
	_getch();
	return 0;
}
```

cpq, 2017-10-26:
FFI can work with up to 6 32-bit arguments and 2 64-bit arguments.
are you on 64-bit windows or on 32-bit?

0x410c, 2017-10-28:
windows is 64but, binary is 32bit compiled, dont remember will post after seeing again, no this direct.h change was not one of them as i remember

0x410c, 2017-11-20:
any update? or support?

cpq, 2017-11-20:
Yes you can.
Take a look at the unit test makefile:
https://github.com/cesanta/mjs/blob/master/mjs/Makefile

It has vc98 and vc2017 targets. mJS is now friendly to vc2017, and to the old vc98 as well.

closed 2017-11-20 by cpq