cesanta/mjs #51: Wrong return value in callback function

engelant, 2017-10-26:
I'm working on STM32F103C8 (Blue-Pill) and am unable to understand the basics of callbacks, at least so it seems since I don't get the expected results.

Here is my C code, just a simple callback test:
```
void (*my_callback)(int, void *);
void *my_user_data;

void timer(int seconds, void (*callback)(int, void *), void *user_data) {
	my_callback = callback;
	my_user_data = user_data;
}

void *my_dlsym(void *handle, const char *name) {
	if (strcmp(name, "timer") == 0)
		return (void *) timer;
	return NULL;
}

int main() {
...
	struct mjs *mjs = mjs_create();
	mjs_set_ffi_resolver(mjs, my_dlsym);
	mjs_exec(mjs, APP_JS, NULL);

	int j = 0;
	while (1) {
		for (uint32_t i = 0; i < 0xFFFFFF; i++) {
			asm("nop");
		}
		my_callback(++j, my_user_data);
	}
}
```

And a little JS:
```
let Timer = {
  set: ffi('void timer(int, void (*)(int, userdata), userdata)')
};

Timer.set(200, function(t) {
  print('Time now: ', t);
}, null);

print('JS is initialized');
```

Console output:
```
JS is initialized 
Time now:  ld  <-- this repeats, but actually should be counting up from 0
```

Also maybe the relevant build options:
```
Building file: ../3rdparty/mjs.c
Invoking: GNU ARM Cross C Compiler
arm-none-eabi-gcc -mcpu=cortex-m3 -mthumb -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -flto  -g -DSTM32F103xB -DMG_ENABLE_FILESYSTEM=0 -DCS_PLATFORM=CS_P_STM32 -DHSE_VALUE=8000000U -I"/home/ivan/eclipse-workspace/blue-squirrel/3rdparty/CMSIS/Include" -I"/home/ivan/eclipse-workspace/blue-squirrel/include" -I"/home/ivan/eclipse-workspace/blue-squirrel/3rdparty/CMSIS/Device/ST/STM32F1xx/Include" -std=gnu11 -MMD -MP -MF"3rdparty/mjs.d" -MT"3rdparty/mjs.o" -c -o "3rdparty/mjs.o" "../3rdparty/mjs.c"
Finished building: ../3rdparty/mjs.c
```
Also I had to remove two lines in the condensed mjs.c file, since it didn't work with them:
```
Line 2200: #include <dirent.h> --> #error "<dirent.h> not supported"
Line 1528: #include <stm32_sdk_hal.h> --> Not using it, not required, I do the low level stuff myself
```

Any help is apprechiated, thx for this great project btw.

cpq, 2017-10-26:
Printing a number turns into https://github.com/cesanta/mjs/blob/master/mjs/src/mjs_util.c#L50-L53  . I presume on your STM32 it turns into `%lld` and your C library does not know how to print `%lld`. E.g. on ESP8266 a newlib C lib is used, and there is a preprocessor directive that turns on support for that kind of stuff, do some research in your case, maybe a similar simple solution exists.

W.r.t. a build - yeah, we need to fix that on our side, thanks

engelant, 2017-10-26:
Exactly, as I'm using newlib too I may have shrinked it down a little with --specs=nano.specs in the linker, what kills support for %lld.

closed 2017-10-26 by engelant