cesanta/mjs #53: Read Access Violation, when calling ffi function on windows

0x410c, 2017-11-21:
Windows 10x64 , 32bit build

Code :
```
#include<windows.h>
#include "mjs.h"
#include <conio.h>

void *my_dlsym(void *handle, const char *name) {
	if (strcmp(name, "msg") == 0) return MessageBoxA;
	return NULL;
}

int WINAPI WinMain(HINSTANCE hInst, HINSTANCE hPrev, LPSTR cmd, int n)
{
	struct mjs *mjs = mjs_create();
	mjs_set_ffi_resolver(mjs, my_dlsym);
	mjs_exec(mjs, "let f = ffi('int msg(int,char*,char*,int)'); f(0,'asdf','asdf',0);", NULL);
	_getch();
	return 0;
}
```
Stack Trace
> >	jsVmTest.exe!mjs_prepend_errorf(mjs * mjs, mjs_err err, const char * fmt, ...) Line 8161	C	Symbols loaded.
>  	jsVmTest.exe!mjs_ffi_call2(mjs * mjs) Line 10484	C	Symbols loaded.
>  	jsVmTest.exe!mjs_execute(mjs * mjs, unsigned int off, unsigned __int64 * res) Line 9275	C	Symbols loaded.
>  	jsVmTest.exe!mjs_exec_internal(mjs * mjs, const char * path, const char * src, int res, unsigned __int64 *) Line 9489	C	Symbols loaded.
>  	[Inline Frame] jsVmTest.exe!mjs_exec(mjs *) Line 9496	C++	Symbols loaded.
>  	jsVmTest.exe!WinMain(HINSTANCE__ * hInst, HINSTANCE__ * hPrev, char * cmd, int n) Line 16	C++	Symbols loaded.
> 

variable values :

> 		err	13	mjs_err
> +		fmt	0x00396e7c "failed to call FFIed function"	const char *
> +		mjs	0x00000002 {bcode_gen={buf=??? len=??? size=??? } bcode_parts={buf=??? len=??? size=??? } bcode_len=...}	mjs *

So after returning fomr the ffi call, mjs object becomes corrupted, as it seems like

0x410c, 2017-11-21:
update

internal function calls are fine,i.e function defined in the c code by the developer,
but calling a winapi function returns a read exception as mjs is corrupted.

cpq, 2017-11-21:
youre very welcome to provide a patch

cpq, 2017-11-21:
by current design, FFI functions should be cdecl.

0x410c, 2017-11-21:
that means the ffi functionality clears the stack after calling the function, which corrupts the stack as messageboxa is using stdcall and has already cleaned up the stack upon returning?

0x410c, 2017-11-21:
and winapi functions are all stdcall?

cpq, 2017-11-21:
yeah, winapi functions are stdcall.
FFI uses cdecl, and all C runtime functions from msvcrtXX.dll are cdecl.

0x410c, 2017-11-21:
any quickfix?

0x410c, 2017-11-21:
ok i compiled it with /Gz flag, which basically convert all function to stdcall, i think this can be a quick fix for now?

cpq, 2017-11-21:
That might work.. although I am not sure about C runtime functions.
Can you successfully FFI stdlib API like `strcmp()` with `/Gz` flag?

0x410c, 2017-11-21:
no i cant i tried, but more or less i want to access winapi directly, everything else can be created as a library, thanks for the help.

closed 2017-11-26 by cpq