cesanta/mjs #54: Multiple Assertion Errors

HongxuChen, 2017-11-23:
We find some assertions errors against mjs with [our developed fuzzing tool FOT](https://github.com/ntu-sec/fot) (will open source in future).

The driver program is mostly copied from the README.md but it accepts an input file:

```c
#include <string.h>
#include <stdio.h>
#include <stdlib.h>
#include "mjs.h"

char* read_file_to_str(char* fname) {
    char * buffer = 0;
    long length;
    FILE * f = fopen(fname, "rb");
    if (f) {
        fseek(f, 0, SEEK_END);
        length = ftell (f);
        fseek(f, 0, SEEK_SET);
        buffer = malloc(length+1);
        if (buffer) {
            fread (buffer, 1, length, f);
        }
        fclose (f);
    }
    buffer[length] = 0;
    return buffer;
}

void foo(int x) {
    printf("Hello %d!\n", x);
}

void *my_dlsym(void *handle, const char *name) {
    if (strcmp(name, "foo") == 0) return foo;
    return NULL;
}

int main(int argc, char** argv) {
    if (argc < 2) {
      printf("usage: %s <js_file>\n", argv[0]);
      exit(1);
    }
    char* content = read_file_to_str(argv[1]);
    // printf("content:\t%s\n", content);
    struct mjs *mjs = mjs_create();
    mjs_set_ffi_resolver(mjs, my_dlsym);
    mjs_err_t res = mjs_exec(mjs, content, NULL);
    // printf("exec res is : %d\n", res);
    mjs_destroy(mjs);
    free(content);
    return res;
}
```

[8399.txt](https://github.com/cesanta/mjs/files/1499517/8399.txt)
> a.out: mjs.c:8399: mjs_val_t mjs_pop_val(struct mbuf *): Assertion `m->len >= sizeof(v)' failed.
> [1]    10370 abort      ../../a.out 8399.txt

[9355.txt](https://github.com/cesanta/mjs/files/1499518/9355.txt)
> a.out: mjs.c:9355: mjs_err_t mjs_execute(struct mjs *, size_t, mjs_val_t *): Assertion `mjs_stack_size(&mjs->scopes) >= scopes_len' failed.
> [1]    25731 abort      ../../a.out 9355.txt

[11829.txt](https://github.com/cesanta/mjs/files/1499519/11829.txt)
> a.out: mjs.c:11829: struct mjs_object *get_object_struct(mjs_val_t): Assertion `mjs_is_object(v)' failed.
> [1]    30905 abort      ../../a.out 11829.txt

cpq, 2017-11-23:
Could you provide a diff to the unit_test.c that fails please?

HongxuChen, 2017-11-23:
@cpq We haven't used unit_test.c, but crafted a bit the driver program (as above). And since we are using fuzzing techniques to generate input files, which are not guaranteed to be valid javascript files (the 3 input above actually contains some invalid bytes).

cpq, 2017-11-23:
So you can capture the inputs that crashes your driver program and put together a failing unit test.
As it is , this report is hard to reproduce, and will remain dormant unless you do what I suggest.

HongxuChen, 2017-11-23:
@cpq We are not quite familiar with the unit testing at this moment. But I think the reproduce of the issue should be straightforward. Simple build and run should be sufficient.

```c
clang-4.0 driver.c mjs.c -ldl -ggdb
./a.out 8399.txt
```

We are not sure whether this is the common use cases for mjs though.

cpq, 2017-11-23:
Got it, there are actually test files attached, thank you.

HongxuChen, 2017-11-24:
There is another assertion error when feeding with `9159.txt`

[9159.txt](https://github.com/cesanta/mjs/files/1501812/9159.txt)

> mjs.out: mjs.c:9159: mjs_err_t mjs_execute(struct mjs *, size_t, mjs_val_t *): Assertion `mjs_stack_size(&mjs->scopes) > 0' failed.
> [1]    30068 abort      ./mjs.out 9159.txt

HongxuChen, 2017-11-24:
In addition, we detected multiple other issues like heap-overflow, stack-overflow, use-after-free errors  when mjs is compiled with sanitizers.

Do you think it proper to open issues for each of them?

cpq, 2017-11-25:
Yes please.

cpq, 2017-11-26:
Thanks.
BTW, I think you can use built-in mjs executable instead of your driver program.
Compile `mjs.c` with `-DMJS_MAIN` flag, and it'll enable a piece of code with `main()` function. The resulting executable can execute files and expressions given by command line flags.

```
$ make build/mjs.exe
docker run -v /Users/lsm/src/cesanta.com:/Users/lsm/src/cesanta.com -w /Users/lsm/src/cesanta.com/mjs docker.cesanta.com/vc98 wine cl mjs.c /DWIN32_LEAN_AND_MEAN /MD /O1 /TC /W2 /WX /I.. /I. /DNDEBUG /DMJS_MEMORY_STATS  -I. -I.. -Isrc -DMJS_MAIN -DMJS_EXPOSE_PRIVATE -DCS_ENABLE_STDIO -DMJS_ENABLE_DEBUG  -D_DARWIN_C_SOURCE /Febuild/mjs.exe
Microsoft (R) 32-bit C/C++ Optimizing Compiler Version 12.00.8168 for 80x86
Copyright (C) Microsoft Corp 1984-1998. All rights reserved.

mjs.c
Microsoft (R) Incremental Linker Version 6.00.8168
Copyright (C) Microsoft Corp 1992-1998. All rights reserved.

/out:build/mjs.exe
mjs.obj
```

HongxuChen, 2017-11-26:
@cpq Thanks for the information, we will follow your advice. May I know what's the expectation of mjs library to handle invalid JS files? We have detected several other heap-buffer-overflow, heap-use-after-free bugs, but most of these are triggered when the input files are not supposed to be well-formed JS (some cannot be interpreted, others even contain illegal bytes that should never occur in JS).

cpq, 2017-11-26:
mjs must handle it as invalid JS - result in syntax error.
what you caught are clearly bugs that have to be fixed.

HongxuChen, 2017-11-26:
@cpq Okay, I see. Please wait for a while for us to provide the additional test inputs that crash, there seems lots of duplicates and we are trying to improving our tool to filter out them.

dimonomid, 2017-12-02:
Thanks for all these reports!

closed 2017-12-02 by dimonomid

HongxuChen, 2017-12-25:
Dear @cpq and @dimonomid as we said previously in this issue and an email sent to @dimonomid a few days before, we provided the test inputs that triggered crashes. There are still some duplicates however it is currently not our priority to remove duplicates with a finer-grained for specific projects.