cesanta/mjs #55: null pointer in mjs_next

HongxuChen, 2017-11-25:
With the driver program mentioned in #54 , our fuzzing tool find a null pointer deference error in `mjs_next`. Below are two sample inputs:

[w03_000043,sig:6,Havoc:1583:3600,src:w01_000567.txt](https://github.com/cesanta/mjs/files/1503132/w03_000043.sig.6.Havoc.1583.3600.src.w01_000567.txt)
[w04_000047,sig:6,Havoc:3273:3440,src:w02_000637.txt](https://github.com/cesanta/mjs/files/1503133/w04_000047.sig.6.Havoc.3273.3440.src.w02_000637.txt)

When compiled with address sanitizer, a sample output will be like:

```
ASAN:DEADLYSIGNAL
=================================================================
==21518==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x00000053b84e bp 0x7fff167ed710 sp 0x7fff167ed6a0 T0)
==21518==The signal is caused by a READ memory access.
==21518==Hint: address points to the zero page.
    #0 0x53b84d  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x53b84d)
    #1 0x5355d9  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x5355d9)
    #2 0x532b51  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x532b51)
    #3 0x5327d3  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x5327d3)
    #4 0x50ba16  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x50ba16)
    #5 0x7fe5ad7e382f  (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #6 0x419b68  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x419b68)
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x53b84d)
==21518==ABORTING
[1]    21518 abort      ../mjs_debug_asan w03_000043,sig:6,Havoc:1583:3600,src:w01_000567.txt
```

A gdb output is like:

```
gdb$ run w03_000043,sig:6,Havoc:1583:3600,src:w01_000567.txt
Starting program: /home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan w03_000043,sig:6,Havoc:1583:3600,src:w01_000567.txt
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Program received signal SIGSEGV, Segmentation fault.
0x000000000053b84e in mjs_next (mjs=0x615000000080, obj=0xfff2611000000320, iterator=0x60d0000000a0) at ../mjs.c:12055
12055       p = ((struct mjs_property *) get_ptr(*iterator))->next;
gdb$ bt
#0  0x000000000053b84e in mjs_next (mjs=0x615000000080, obj=0xfff2611000000320, iterator=0x60d0000000a0) at ../mjs.c:12055
#1  0x00000000005355da in mjs_execute (mjs=0x615000000080, off=0x0, res=0x7fffffffb7c0) at ../mjs.c:9193
#2  0x0000000000532b52 in mjs_exec_internal (mjs=0x615000000080, path=0x5892e0 <.str> "<stdin>", src=0x623000000100 "let s = '', sum = 0, o = {a: 1, b: 3, c: 3};\nfor (let k in o) {\n\ts = s + k;\n\tsum += 0, o = {a: 1, b: 3, c: 3};\nfor (let k in o) {\n\ts = s + k;\n\tsum &= o[k];\n}\n\n// \"\"'", '\260' <repeats 14 times>, " = 0, o = {a: 1, b: 3"..., generate_jsc=0x0, res=0x0) at ../mjs.c:9489
#3  0x00000000005327d4 in mjs_exec (mjs=0x615000000080, src=0x623000000100 "let s = '', sum = 0, o = {a: 1, b: 3, c: 3};\nfor (let k in o) {\n\ts = s + k;\n\tsum += 0, o = {a: 1, b: 3, c: 3};\nfor (let k in o) {\n\ts = s + k;\n\tsum &= o[k];\n}\n\n// \"\"'", '\260' <repeats 14 times>, " = 0, o = {a: 1, b: 3"..., res=0x0) at ../mjs.c:9496
#4  0x000000000050ba17 in main (argc=0x2, argv=0x7fffffffba58) at ../driver.c:42
```

HongxuChen, 2017-11-25:
There are another two inputs that will cause segfaults possibly resulting from the same root cause, despite that the invalid read addresses are different.
[w03_000045,sig:6,Havoc:240:261,src:w02_000502.txt](https://github.com/cesanta/mjs/files/1503223/w03_000045.sig.6.Havoc.240.261.src.w02_000502.txt)
[w04_000020,sig:6,Havoc:11739:21568,src:w01_000299.txt](https://github.com/cesanta/mjs/files/1503224/w04_000020.sig.6.Havoc.11739.21568.src.w01_000299.txt)

When compiled with address sanitizer, it gives the outputs:

w03_000045,sig:6,Havoc:240:261,src:w02_000502.txt
```
ASAN:DEADLYSIGNAL
=================================================================
==1648==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000044 (pc 0x00000053b84e bp 0x7fff7b2b04d0 sp 0x7fff7b2b0460 T0)
==1648==The signal is caused by a READ memory access.
==1648==Hint: address points to the zero page.
    #0 0x53b84d  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x53b84d)
    #1 0x5355d9  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x5355d9)
    #2 0x532b51  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x532b51)
    #3 0x5327d3  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x5327d3)
    #4 0x50ba16  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x50ba16)
    #5 0x7f959be6682f  (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #6 0x419b68  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x419b68)
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x53b84d)
==1648==ABORTING
```

w04_000020,sig:6,Havoc:11739:21568,src:w01_000299.txt
```
ASAN:DEADLYSIGNAL
=================================================================
==1661==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000042 (pc 0x00000053b84e bp 0x7fff4794f850 sp 0x7fff4794f7e0 T0)
==1661==The signal is caused by a READ memory access.
==1661==Hint: address points to the zero page.
    #0 0x53b84d  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x53b84d)
    #1 0x5355d9  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x5355d9)
    #2 0x532b51  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x532b51)
    #3 0x5327d3  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x5327d3)
    #4 0x50ba16  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x50ba16)
    #5 0x7f5fd11cd82f  (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #6 0x419b68  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x419b68)
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x53b84d)
```

dimonomid, 2017-12-01:
Resolved with https://github.com/cesanta/mjs/commit/db0e2d696be94df098ca42e1e417cf0c500014a0

closed 2017-12-01 by dimonomid