cesanta/mjs #56: null pointer in mjs_execute

HongxuChen, 2017-11-25:
With the driver program mentioned in #54 , our fuzzing tool find a null pointer deference error in mjs_next. Below are two sample inputs:

[w03_000054,sig:6,Splice:1:1,src:w04_000548.txt](https://github.com/cesanta/mjs/files/1503140/w03_000054.sig.6.Splice.1.1.src.w04_000548.txt)
[w04_000038,sig:6,Splice:8:24,src:w02_000173.txt](https://github.com/cesanta/mjs/files/1503141/w04_000038.sig.6.Splice.8.24.src.w02_000173.txt)

When compiled with address sanitizer, an output for the first input file will be like:

```
ASAN:DEADLYSIGNAL
=================================================================
==25622==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x00000053681e bp 0x7ffe6f193050 sp 0x7ffe6f192760 T0)
==25622==The signal is caused by a READ memory access.
==25622==Hint: address points to the zero page.
    #0 0x53681d  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x53681d)
    #1 0x532b51  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x532b51)
    #2 0x5327d3  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x5327d3)
    #3 0x50ba16  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x50ba16)
    #4 0x7f68416e682f  (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #5 0x419b68  (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x419b68)
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan+0x53681d)
==25622==ABORTING
```

a gdb output:

```
gdb$ run w03_000054,sig:6,Splice:1:1,src:w04_000548.txt
Starting program: /home/hongxu/FUSE/tests/mjs/github/mjs_debug_asan w03_000054,sig:6,Splice:1:1,src:w04_000548.txt
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Program received signal SIGSEGV, Segmentation fault.
0x000000000053681e in mjs_execute (mjs=0x615000000080, off=0x0, res=0x7fffffffb7c0) at ../mjs.c:9354
9354            size_t scopes_len = mjs_get_int(mjs, *vptr(&mjs->loop_addresses, -3));
gdb$ bt
#0  0x000000000053681e in mjs_execute (mjs=0x615000000080, off=0x0, res=0x7fffffffb7c0) at ../mjs.c:9354
#1  0x0000000000532b52 in mjs_exec_internal (mjs=0x615000000080, path=0x5892e0 <.str> "<stdin>", src=0x623000000100 "let o = {a: 1, 'b': 2, c: {d: 3}, d: [1, 2, 3, 'xx']};\nlet a = 'foo', b = 'bar';\nlet s0 = 'a' + \"b\";\nlet s2 = 'abcdef' + \"123456\";\n1 - 2 + 3 + o['a']  continue;\n  e \253= d;\n  break;\n}\n\na === 45 &&&& typ"..., generate_jsc=0x0, res=0x0) at ../mjs.c:9489
#2  0x00000000005327d4 in mjs_exec (mjs=0x615000000080, src=0x623000000100 "let o = {a: 1, 'b': 2, c: {d: 3}, d: [1, 2, 3, 'xx']};\nlet a = 'foo', b = 'bar';\nlet s0 = 'a' + \"b\";\nlet s2 = 'abcdef' + \"123456\";\n1 - 2 + 3 + o['a']  continue;\n  e \253= d;\n  break;\n}\n\na === 45 &&&& typ"..., res=0x0) at ../mjs.c:9496
#3  0x000000000050ba17 in main (argc=0x2, argv=0x7fffffffba68) at ../driver.c:42
```

dimonomid, 2017-12-02:
Fixed, thanks!

closed 2017-12-02 by dimonomid

attritionorg, 2018-02-20:
@dimonomid Can you link to the fixing commit? Not seeing any immediately obvious match-ups.