cesanta/mjs #59: divide-by-zero error in do_arith_op

HongxuChen, 2017-11-25:
With the driver program mentioned in #54 , our fuzzing tool FOT found a divide-by-zero error in `do_arith_op`.

[w01_000002,sig:6,Arith8:286:486980,src:w02_000513.txt](https://github.com/cesanta/mjs/files/1503237/w01_000002.sig.6.Arith8.286.486980.src.w02_000513.txt)

```
Program received signal SIGFPE, Arithmetic exception.
0x000000000041e3c1 in do_arith_op (da=0, db=0.20000000000000001, op=0xf) at ../mjs.c:8603
8603            return (double) ((int64_t) da % (int64_t) db);
gdb$ bt
#0  0x000000000041e3c1 in do_arith_op (da=0, db=0.20000000000000001, op=0xf) at ../mjs.c:8603
#1  0x000000000041df5b in do_op (mjs=0x62b010, a=0x0, b=0x3fc999999999999a, op=0xf) at ../mjs.c:8649
#2  0x000000000041c2a5 in exec_expr (mjs=0x62b010, op=0xf) at ../mjs.c:8719
#3  0x000000000040e5fb in mjs_execute (mjs=0x62b010, off=0x0, res=0x7fffffffb8d8) at ../mjs.c:9314
#4  0x000000000040cf4e in mjs_exec_internal (mjs=0x62b010, path=0x4248e7 "<stdin>", src=0x62c250 "0%1%0.2\"';\n\tnts of this file are subject to the Netscape Public\n * License Version 1.1 (the \"Lic", generate_jsc=0x0, res=0x0) at ../mjs.c:9489
#5  0x000000000040ce94 in mjs_exec (mjs=0x62b010, src=0x62c250 "0%1%0.2\"';\n\tnts of this file are subject to the Netscape Public\n * License Version 1.1 (the \"Lic", res=0x0) at ../mjs.c:9496
#6  0x00000000004015e5 in main (argc=0x2, argv=0x7fffffffba68) at ../driver.c:4
```

The `if (db != 0)` at line 8602 should not be used for `double` type.

cpq, 2017-11-26:
Should the attached test file supposed be a valid JS ?
It does not appear to be.

HongxuChen, 2017-11-26:
@cpq It's not a valid JS. And the other test input files are mostly not valid JS. 

We suppose these are not severe bugs, but probably it's better that mjs exits with non-zero code rather than aborts.

dimonomid, 2017-12-01:
Resolved by https://github.com/cesanta/mjs/commit/618eb4241f28444e44da1a5122493bed29465be7

closed 2017-12-01 by dimonomid