cesanta/mjs #61: Main thread blocks when a parsing error is present

ovidiugabriel, 2017-12-14:
I have some troubles running the  **mjs**. If I have an invalid syntax in the .js files, the main thread halts.

```c++
WINAPI WinMain(HINSTANCE, HINSTANCE, LPSTR args, int)
{
    mjs = mjs_create();
    mjs_set_ffi_resolver(mjs, resolve_symbol);

    mjs_err_t errcd = mjs_exec_file(mjs, "main.js", NULL);
    if (MJS_OK != errcd) {
        LOG << mjs_strerror(mjs, errcd);
        abort();
    }

    printf("mjs done"); fflush(stdout);
    // ...
}

How can I fix this?

dimonomid, 2018-01-26:
And what does your `main.js` look like? I tried just this:
```
sdf
```
and I get the expected `[sdf] is not defined`.

closed 2018-02-09 by dimonomid

ovidiugabriel, 2018-07-23:
Sorry, that was another version of the code. Here is the actual problem.

```cpp
   // Here we start the thread (this is in WinMain)
unitTestThreadHandle = CreateThread( NULL, 0, mjs_thread, NULL, 0, NULL);
```

If main.js has any syntax error, the mjs_thread is blocked in executing mjs_exec_file().
So "[mjs] mjs_exec_file() executed " is never printed.

```cpp
unsigned long __stdcall mjs_thread( void* arg ) {
    LOG << "[mjs] Starting thread";

    mjs = mjs_create();
    mjs_set_ffi_resolver(mjs, resolve_symbol);

    if (mjs) {
        mjs_err_t errcd = mjs_exec_file(mjs, "main.js", NULL);
        LOG << "[mjs] mjs_exec_file() executed ";
        if (MJS_OK != errcd) {
            LOG << "[mjs] Internal error while executing 'main.js'";
        } else {
            LOG << "[mjs] main.js execution was fine";
        }
    }

    LOG << "[mjs] Thread exited";

    return 0;
}
```

For a better handling we registered `atexit(onExit);`. That prints also for evidence that msj thread doesn't exit.

```cpp
void onExit(void)
{
    LOG << "[mjs] Wait for mjs thread";
    DWORD errcd = WaitForSingleObject(unitTestThreadHandle, 1000 /* ms */);
    switch (errcd) {
        case WAIT_TIMEOUT:
        {
            LOG << "[mjs] Timeout waiting for mjs thread to close. "
                "Possible JavaScript syntax error.";
        }
        break;
    }
}
```

How can we register a callback to catch errors and print them to a file? I don't know why the thread is blocking but I assume that mjs will be able to call an error handler callback even if the mjs_exec_file() does not return to the caller.

ovidiugabriel, 2018-07-23:
I enabled `#define CS_ENABLE_DEBUG 1`, maybe this helps.
```
.\mjs.c:11920          PNEXT 11920
.\mjs.c:13703          --> [s]
.\mjs.c:12653        [s[df]
.\mjs.c:12035        [s] 03C6FC4C
.\mjs.c:13703          --> [[]
.\mjs.c:12102          PNEXT 12102
.\mjs.c:13703          --> [[]
.\mjs.c:12113          PNEXT 12113
.\mjs.c:13703          --> [df]
.\mjs.c:12035        [df] 03C6FA2C
.\mjs.c:13703          --> []
.\mjs.c:12102          PNEXT 12102
.\mjs.c:13703          --> []

```