cesanta/mjs #62: Docker dependency

Mazeto, 2017-12-17:
Please,

    add an option to compile without docker.

Thanks.

lpsantil, 2017-12-20:
I second Mazeto's request.  It's an odd requirement to need Docker installed to compile a JS Engine intended for Microcontrollers.  I get that it makes the build process fixed and consistent.  But, as someone where my day job is working on OpenShift, Kubernetes, and Docker, I find the decision odd.

Wouldn't be better to package mjs as a Container itself?  The use case calls for mjs, not building mjs.

jnz86, 2017-12-20:
I agree.

davidbuzz, 2018-01-16:
this worked for me on OSX to build it without docker: 
cd mjs/mjs
clang -lm -ldl -std=c99 -Wall -Wextra -pedantic -g  -I. -I.. -Isrc -DMJS_MAIN -DMJS_EXPOSE_PRIVATE -DCS_ENABLE_STDIO -DMJS_ENABLE_DEBUG  -D_DARWIN_C_SOURCE -DCS_MMAP  src/../../common/cs_dbg.c src/../../common/cs_file.c src/../../common/cs_varint.c src/../../common/mbuf.c src/../../common/mg_str.c src/../../common/str_util.c src/../../frozen/frozen.c src/ffi/ffi.c src/mjs_array.c src/mjs_bcode.c src/mjs_builtin.c src/mjs_conversion.c src/mjs_core.c src/mjs_dataview.c src/mjs_exec.c src/mjs_ffi.c src/mjs_gc.c src/mjs_json.c src/mjs_main.c src/mjs_object.c src/mjs_parser.c src/mjs_primitive.c src/mjs_string.c src/mjs_tok.c src/mjs_util.c -o build/mjs

Mazeto, 2018-01-23:
@davidbuzz Thx.

lukeb-uk, 2018-02-02:
I have mJS as a container here if it's useful to anyone: https://hub.docker.com/r/lukebuk/mjs/

cpq, 2018-05-08:
```
make clean all DOCKER_CLANG=
``