cesanta/mjs #65: stackoverflow error

HongxuChen, 2017-12-25:
The follow input file will cause a stackoverflow when mjs is compiled with address sanitizer.

[w01_000006,sig:6,Splice:10:16,src:w01_000170.txt](https://github.com/cesanta/mjs/files/1585626/w01_000006.sig.6.Splice.10.16.src.w01_000170.txt)

```
/home/hongxu/FUSE/tests/mjs-ins/out_20171215_01_12_01/crash/w01_000006,sig:6,Splice:10:16,src:w01_000170
ASAN:DEADLYSIGNAL
=================================================================
==21078==ERROR: AddressSanitizer: stack-overflow on address 0x7ffecfdbbf78 (pc 0x0000004371aa bp 0x7ffecfdbc7f0 sp 0x7ffecfdbbf80 T0)
    #0 0x4371a9  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x4371a9)
    #1 0x5532c1  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x5532c1)
    #2 0x5517d9  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x5517d9)
    #3 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #4 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #5 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #6 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #7 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #8 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #9 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #10 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #11 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #12 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #13 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #14 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #15 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #16 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #17 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #18 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #19 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #20 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #21 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #22 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #23 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #24 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #25 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #26 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #27 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #28 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #29 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #30 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #31 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #32 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #33 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #34 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #35 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #36 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #37 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #38 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #39 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #40 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #41 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #42 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #43 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #44 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #45 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #46 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #47 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #48 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #49 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #50 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #51 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #52 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #53 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #54 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #55 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #56 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #57 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #58 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #59 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #60 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
[... 169 lines cut ...]
    #230 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #231 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #232 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #233 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #234 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #235 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #236 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #237 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #238 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #239 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #240 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #241 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #242 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #243 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #244 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #245 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #246 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #247 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)
    #248 0x551940  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x551940)
    #249 0x55e7f7  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55e7f7)
    #250 0x55619e  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x55619e)

SUMMARY: AddressSanitizer: stack-overflow (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x4371a9) 
==21078==ABORTING
```

HongxuChen, 2017-12-25:
[w01_000022,sig:6,Splice:5:17,src:w04_000638.txt](https://github.com/cesanta/mjs/files/1585649/w01_000022.sig.6.Splice.5.17.src.w04_000638.txt)

dimonomid, 2018-01-26:
Failed to reproduce. The command to build mjs was:
```
gcc -fsanitize=address -lm -std=c99 -Wall -Wextra -pedantic -g  -I. -I.. -Isrc -DMJS_MAIN -DMJS_EXPOSE_PRIVATE -DCS_ENABLE_STDIO -DMJS_ENABLE_DEBUG  -DCS_MMAP -Wl,--no-as-needed -ldl  src/../../common/cs_dbg.c src/../../common/cs_file.c src/../../common/cs_varint.c src/../../common/mbuf.c src/../../common/mg_str.c src/../../common/str_util.c src/../../frozen/frozen.c src/ffi/ffi.c src/mjs_array.c src/mjs_bcode.c src/mjs_builtin.c src/mjs_conversion.c src/mjs_core.c src/mjs_dataview.c src/mjs_exec.c src/mjs_ffi.c src/mjs_gc.c src/mjs_json.c src/mjs_main.c src/mjs_object.c src/mjs_parser.c src/mjs_primitive.c src/mjs_string.c src/mjs_tok.c src/mjs_util.c -o build/mjs
```
And running the provided input gives correct parse errors, e.g. this is the result of running the first provided input:
```
$ build/mjs /var/tmp/t1
MJS error: parse error at line 1: [$$$$$$$$$]
```

HongxuChen, 2018-01-26:
Hi, here is the compilation options I used:`clang ../mjs-src/mjs.c -fsanitize=address -lm -std=c99 -Wall -Wextra -pedantic -g -DMJS_MAIN -O3`.

```
$ clang --version
clang version 4.0.0-1ubuntu1~16.04.2 (tags/RELEASE_400/rc1)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

dimonomid, 2018-01-26:
Thanks, reproduced, will be fixed soon.

closed 2018-01-26 by rojer

dimonomid, 2018-01-26:
Fixed in the private repo, will be published here soon.