cesanta/mjs #66: SEGV on unknown address

HongxuChen, 2017-12-25:
The input file below will trigger crash when mjs is compiled with address sanitizer.

[w01_000017,sig:6,Splice:1:1,src:w03_000633.txt](https://github.com/cesanta/mjs/files/1585633/w01_000017.sig.6.Splice.1.1.src.w03_000633.txt)

```
ASAN:DEADLYSIGNAL
=================================================================
==31939==ERROR: AddressSanitizer: SEGV on unknown address 0x00000053d0d5 (pc 0x00000053d0d5 bp 0x7ffe819f08b0 sp 0x7ffe819eff78 T0)
==31939==The signal is caused by a WRITE memory access.
    #0 0x53d0d4 in mjs_load /home/hongxu/FUSE/tests/mjs-ins/../mjs/mjs.c:8061:1
    #1 0x54357c in mjs_exec_internal /home/hongxu/FUSE/tests/mjs-ins/../mjs/mjs.c:9706:5
    #2 0x5439a8 in mjs_exec_file /home/hongxu/FUSE/tests/mjs-ins/../mjs/mjs.c:9729:11
    #3 0x54c981 in main /home/hongxu/FUSE/tests/mjs-ins/../mjs/mjs.c:12009:11
    #4 0x7f3348e4282f in __libc_start_main /build/glibc-bfm8X4/glibc-2.23/csu/../csu/libc-start.c:291
    #5 0x419e88 in _start (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT-new.out+0x419e88)
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/hongxu/FUSE/tests/mjs-ins/../mjs/mjs.c:8061:1 in mjs_load
```

gdb info:

```
gdb$ run crash/w01_000017,sig:6,Splice:1:1,src:w03_000633
Starting program: /home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT-new.out crash/w01_000017,sig:6,Splice:1:1,src:w03_000633
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Program received signal SIGSEGV, Segmentation fault.
0x000000000053d0d5 in mjs_load (mjs=0x20) at ../mjs/mjs.c:8061
8061    }
gdb$ bt
#0  0x000000000053d0d5 in mjs_load (mjs=0x20) at ../mjs/mjs.c:8061
#1  0x000000000054357d in mjs_exec_internal (mjs=<error reading variable: Cannot access memory at address 0xf1>, path=<error reading variable: Cannot access memory at address 0xe9>, src=<error reading variable: Cannot access memory at address 0xe1>, generate_jsc=<error reading variable: Cannot access memory at address 0xdd>, res=<error reading variable: Cannot access memory at address 0xd1>) at ../mjs/mjs.c:9706
#2  0x00000000005439a9 in mjs_exec_file (mjs=0x615000000080, path=0x7fffffffc080 "crash/w01_000017,sig:6,Splice:1:1,src:w03_000633", res=0x7fffffffb6c0) at ../mjs/mjs.c:9729
#3  0x000000000054c982 in main (argc=0x2, argv=0x7fffffffb958) at ../mjs/mjs.c:12009
```

HongxuChen, 2017-12-25:
[w02_000023,sig:6,Splice:1:1,src:w01_000651.txt](https://github.com/cesanta/mjs/files/1585666/w02_000023.sig.6.Splice.1.1.src.w01_000651.txt)

dimonomid, 2018-01-26:
This is a flipside of the mJS <-> C interface: it is possible to write mJS code which would "shoot yourself in the foot". This code basically does this:
```
--print;
print()
```
Where `print` is a C-function, i.e. it's just an address in memory. mJS allows pointer arithmetics, so `--print` effectively decrements the function address, and `print()` calls a function with an invalid address, which obviously leads to crash.

dimonomid, 2018-01-26:
Another "canonical" way to shoot yourself in the foot:
```
let p = ffi('void *malloc(int)')(16);
p[16] = 0x12;
```
(to get the guaranteed crash this, address sanitizer is required of course)

closed 2018-01-26 by dimonomid