cesanta/mjs #67: random results with different versions/compilation options

HongxuChen, 2017-12-25:
The following file will cause different results when compiled with -O0/-O3, and will cause segfault with prior versions.

[w01_000020,sig:4,Splice:1:1,src:w01_000648.txt](https://github.com/cesanta/mjs/files/1585644/w01_000020.sig.4.Splice.1.1.src.w01_000648.txt)

```
mjs-ins/out_20171215_01_12_01 > ../mjs_debug_main_FOT-new.out /home/hongxu/FUSE/tests/mjs-ins/out_20171215_01_12_01/crash/w01_000020,sig:4,Splice:1:1,src:w01_000648                                      < 5:16 PM
teststatu\x8fction tas );EEEEEEEEEEEE\x0astatu\x8fction tas );EEEEEEEEEEEE\x0a=tas );EEEEEEEEEEEE\x0a=\x0aZ;;;;;;;t;tas
teststatu\x8fction tas );EEEEEEEEEEEE\x0a=tas );EEEEE );EEEEEEEEEEEE\x0a=tas );EEEEE\x0aEEEEEEEEEEE\x0a=\x0aZ;;;;;;;t;tas
teststatu\x8fction tas );EEEEEEEEEEEE\x0astatu\x8fction tas );EEEEEEEEEEEE\x0a=tas );EEEEEEEEEEEE\x0a=\x0aZ;;;;;;;t;tas
teststatu\x8fction tas );EEEEEEEEEEEE\x0a=tas );EEEEE );EEEEEEEEEEEE\x0a=tas );EEEEEEEEEEEE\x0a=\x0aZ;;;;;;;t;tas
teas );EEEEEEEEEEEE\x0a=tas );EEEEEEEEEEEE\x0a=\x0aZ;;;;;;;t;tas
teststatu\x8fction tas );EEEEEEEEEEEE\x0a=tas );EEEEE );EEEEEEEEEEEE\x0a=tas );EEEEEEEEEEEE\x0a=\x0aZ;;;;;;;t;tas
teststatu\x8fctio;tas
teststatu\x8fction tas );EEEEEEEEEEEE\x0astatu\x8fction tas );EEEEEEEEEEEE\x0a=tas );EEEEEEEEEEEE\x0a=\x0aZ;;;;;;;t;tas
teststatu\x8fction tas );EEEEEEEEEEEE\x0a=tas );EEEEE );EEEEEEEEEEEE\x0a=tas );EEEEE\x0aEEEEEEEEEEE\x0a=\x0aZ;;;;;;;t;tas
teststatu\x8fction tas );EEEEEEEEEEEE\x0astatu\x8fction tas );EEEEEEEEEEEE\x0a=tas );EEEEEEEEEEEE\x0a=\x0aZ;;;;;;;t;tas
teststatu\x8fction tas );EEEEEEEEEEEE\x0a=tas );EEEEE );EEEEEEEEEEEE\x0a=tas );EEEEEEEEEEEE\x0a=\x0aZ;;;;;;;t;tas
teas );EEEEEEEEEEEE\x0a=tas );EEEEEEEEEEEE\x0a=\x0aZ;;;;;;;t;tas
teststatu\x8fction tas );EEEEEEEEEEEE\x0a=tas );EEEEE );EEEEEEEEEEEE\x0a=tas );EEEEEEEEEEEE\x0a=\x0aZ;;;;;;;t;tas
teststatu\x8fction tas );EEEEEEEEEEEE\x0a=tas =;EEEEEEEEEEEE\x0a=\x0aZ;;;;;;;t;tas
undefined
mjs-ins/out_20171215_01_12_01 > less /home/hongxu/FUSE/tests/mjs-ins/out_20171215_01_12_01/crash/w01_000020,sig:4,Splice:1:1,src:w01_000648                                                               < 5:16 PM
mjs-ins/out_20171215_01_12_01 > ../mjs_debug_main_FOT.out /home/hongxu/FUSE/tests/mjs-ins/out_20171215_01_12_01/crash/w01_000020,sig:4,Splice:1:1,src:w01_000648                                          < 5:16 PM
[1]    22548 illegal hardware instruction  ../mjs_debug_main_FOT.out
```

dimonomid, 2018-01-26:
Same as in https://github.com/cesanta/mjs/issues/66 : the provided input adjusts `print`, a plain C function pointer, and then calls it. Obviously, it will result in a crash. In mJS, one is allowed to shoot himself in the foot.

Probably we could make mjs to distinguish between kinds of pointers and disallow adjusting pointers to functions, however not sure it'll actually make things a lot better. Let me discuss that with the team.

dimonomid, 2018-01-26:
At the moment the decision was not to distinguish between kinds of pointers.

closed 2018-01-26 by dimonomid