cesanta/mjs #68: illegal write to unknown address

HongxuChen, 2017-12-25:
input file:

[w01_000021,sig:6,Havoc:46:153,src:w04_000614.txt](https://github.com/cesanta/mjs/files/1585646/w01_000021.sig.6.Havoc.46.153.src.w04_000614.txt)

```
/home/hongxu/FUSE/tests/mjs-ins/out_20171215_01_12_01/crash/w01_000021,sig:6,Havoc:46:153,src:w04_000614
ASAN:DEADLYSIGNAL
=================================================================
==23110==ERROR: AddressSanitizer: SEGV on unknown address 0x000000526dae (pc 0x000000526dae bp 0x7ffe95877730 sp 0x7ffe958771d8 T0)
==23110==The signal is caused by a WRITE memory access.
    #0 0x526dad  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x526dad)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT.out+0x526dad) 
==23110==ABORTING
```

HongxuChen, 2017-12-25:
[w02_000014,sig:6,Splice:17:21,src:w01_000618.txt](https://github.com/cesanta/mjs/files/1585660/w02_000014.sig.6.Splice.17.21.src.w01_000618.txt)

dimonomid, 2018-01-26:
Same issue with `print` as here https://github.com/cesanta/mjs/issues/66

closed 2018-01-26 by dimonomid