cesanta/mjs #69: illegal read from unknown address

HongxuChen, 2017-12-25:
input file:
[w02_000018,sig:6,Splice:1:1,src:w01_000651.txt](https://github.com/cesanta/mjs/files/1585662/w02_000018.sig.6.Splice.1.1.src.w01_000651.txt)

```
ASAN:DEADLYSIGNAL
=================================================================
==4480==ERROR: AddressSanitizer: SEGV on unknown address 0x00000053d0d0 (pc 0x00000053d0d0 bp 0x7ffce3e6eb10 sp 0x7ffce3e6e1d8 T0)
==4480==The signal is caused by a WRITE memory access.
    #0 0x53d0cf in mjs_load /home/hongxu/FUSE/tests/mjs-ins/../mjs/mjs.c:8061:1
    #1 0x54357c in mjs_exec_internal /home/hongxu/FUSE/tests/mjs-ins/../mjs/mjs.c:9706:5
    #2 0x5439a8 in mjs_exec_file /home/hongxu/FUSE/tests/mjs-ins/../mjs/mjs.c:9729:11
    #3 0x54c981 in main /home/hongxu/FUSE/tests/mjs-ins/../mjs/mjs.c:12009:11
    #4 0x7fb6e49ff82f in __libc_start_main /build/glibc-bfm8X4/glibc-2.23/csu/../csu/libc-start.c:291
    #5 0x419e88 in _start (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT-new.out+0x419e88)
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/hongxu/FUSE/tests/mjs-ins/../mjs/mjs.c:8061:1 in mjs_load
==4480==ABORTING
```

HongxuChen, 2017-12-25:
[w02_000030,sig:6,Splice:1:1,src:w03_000685.txt](https://github.com/cesanta/mjs/files/1585671/w02_000030.sig.6.Splice.1.1.src.w03_000685.txt)

attritionorg, 2017-12-25:
"The signal is caused by a WRITE memory access."

dimonomid, 2018-01-26:
The same thing with `print` function https://github.com/cesanta/mjs/issues/66

closed 2018-01-26 by dimonomid