cesanta/mjs #72: illegal write

HongxuChen, 2017-12-25:
[w03_000032,sig:6,Splice:14:16,src:w04_000692.txt](https://github.com/cesanta/mjs/files/1585702/w03_000032.sig.6.Splice.14.16.src.w04_000692.txt)

```
w03_000032,sig:6,Splice:14:16,src:w04_000692
undefined 
ASAN:DEADLYSIGNAL
=================================================================
==28523==ERROR: AddressSanitizer: SEGV on unknown address 0x00000053d0ee (pc 0x00000053d0ee bp 0x7fff92450330 sp 0x7fff9244fb58 T0)
==28523==The signal is caused by a WRITE memory access.
    #0 0x53d0ed  (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT-new.out+0x53d0ed)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/home/hongxu/FUSE/tests/mjs-ins/mjs_debug_main_FOT-new.out+0x53d0ed) 
==28523==ABORTING
```

HongxuChen, 2017-12-25:
[w03_000034,sig:6,Havoc:334:1024,src:w04_000697.txt](https://github.com/cesanta/mjs/files/1585703/w03_000034.sig.6.Havoc.334.1024.src.w04_000697.txt)

dimonomid, 2018-01-26:
The same problem with `print--` https://github.com/cesanta/mjs/issues/66

closed 2018-01-26 by dimonomid