cesanta/mjs #80: RPC.call(RPC.LOCAL,'FS.List'): JSON.parse() fails in api_rpc.js

Spiritdude, 2018-01-26:
I get strange parsing errors after RPC.call(...,"FS.List") to retrieve file list (there seems no File.list()):

The data in question is:
```
["api_net.jsc","api_dataview.js","api_wifi.js","api_pwm.js","api_i2c.js","api_mqtt.jsc","api_esp8266.js","api_file.js","api_config.jsc","ca.pem","api_aws.js","api_math.js","api_net.js","api_adc.js","api_events.js","api_bitbang.js","mgos_config_schema.json","api_sys.js","api_neopixel.js","api_uart.js","api_mqtt.js","api_config.js","api_dht.js","conf0.json","api_grove.js","api_ota.js","api_shadow.js","hello.js","hello.jsc","api_arch_uart.js","api_spi.js","api_http.js","api_log.js","api_events.jsc","api_gpio.jsc","api_timer.js","conf9.json","api_wifi.jsc","api_file.jsc","init.js","init.jsc","api_rpc.js","api_rpc.jsc","ls.js","ls.jsc","mgos_ro_vars_schema.json","api_sys.jsc","api_timer.jsc","api_math.jsc","api_gpio.js"]
```
which seems `JSON.parse()` cannot parse and fails:
```
..... at api_rpc.js:21
[Jan 19 23:37:18.718] MJS callback error: invalid JSON string
```
`api_rpc.js`:
```
20:  _ccb: function(res, code, msg, ud) {
21:    ud.cb(res ? JSON.parse(res) : null, code, msg, ud.ud);
22:    ffi_cb_free(RPC._ccb, ud);
23:  },
```
If I try this in `init.js`:
```
let s = '["api_net.jsc","api_dataview.js","api_wifi.js","api_pwm.js","api_i2c.js","api_mqtt.jsc","api_esp8266.js","api_file.js","api_config.jsc","ca.pem","api_aws.js","api_math.js","api_net.js","api_adc.js","api_events.js","api_bitbang.js","mgos_config_schema.json","api_sys.js","api_neopixel.js","api_uart.js","api_mqtt.js","api_config.js","api_dht.js","conf0.json","api_grove.js","api_ota.js","api_shadow.js","hello.js","hello.jsc","api_arch_uart.js","api_spi.js","api_http.js","api_log.js","api_events.jsc","api_gpio.jsc","api_timer.js","conf9.json","api_wifi.jsc","api_file.jsc","init.js","init.jsc","api_rpc.js","api_rpc.jsc","ls.js","ls.jsc","mgos_ro_vars_schema.json","api_sys.jsc","api_timer.jsc","api_math.jsc","api_gpio.js"]';
JSON.parse(s);
```
it fails on my ESP8266 (Wemos D1 Mini) with
```
[Jan 19 23:45:57.256]   at init.js:35
[Jan 19 23:45:57.256] MJS error: invalid JSON string
```
The string is 725 bytes long.

I truncated the string by apprx. half of the array, then `JSON.parse()` does not throw an error.
```
print("hello world");
let s0 = '["api_net.jsc","api_dataview.js","api_wifi.js","api_pwm.js","api_i2c.js","api_mqtt.jsc","api_esp8266.js"]';
JSON.parse(s0);
let s1 = '["api_net.jsc","api_dataview.js","api_wifi.js","api_pwm.js","api_i2c.js","api_mqtt.jsc","api_esp8266.js","api_file.js","api_config.jsc","ca.pem","api_aws.js","api_math.js","api_net.js","api_adc.js","api_events.js","api_bitbang.js","mgos_config_schema.json"]';
JSON.parse(s1);
let s2 = '["api_net.jsc","api_dataview.js","api_wifi.js","api_pwm.js","api_i2c.js","api_mqtt.jsc","api_esp8266.js","api_file.js","api_config.jsc","ca.pem","api_aws.js","api_math.js","api_net.js","api_adc.js","api_events.js","api_bitbang.js","mgos_config_schema.json","api_sys.js","api_neopixel.js","api_uart.js","api_mqtt.js","api_config.js","api_dht.js","conf0.json","api_grove.js","api_ota.js","api_shadow.js","hello.js","hello.jsc","api_arch_uart.js","api_spi.js","api_http.js","api_log.js","api_events.jsc","api_gpio.jsc","api_timer.js","conf9.json","api_wifi.jsc","api_file.jsc","init.js","init.jsc","api_rpc.js","api_rpc.jsc","ls.js","ls.jsc","mgos_ro_vars_schema.json","api_sys.jsc","api_timer.jsc","api_math.jsc","api_gpio.js"]';
JSON.parse(s2);
```
it fails with `s2`, but not `s0` or `s1`.

I tried to make a loop and push elements in an array and stringify and parse it again to find the limit, but brought me in a boot loop . . .

Spiritdude, 2018-01-26:
The same code with ESP32 (Wemos LOLIN32 Lite) JSON.parse() works.

dimonomid, 2018-01-29:
Thanks for reporting, right, there was a bug in `JSON.parse` when JSON contains a number of strings enough to reallocate internal strings buffer, which caused the source string to be invalidated. It is fixed in the internal repo, will be published here soon.