cesanta/mjs #81: Questionable return value behavior on nrf52 / cortex m4

jnz86, 2018-01-26:
struct mjs *mjs = mjs_create();
mjs_val_t ret=0;
mjs_exec(mjs, "let ret=0;", &ret);

Here are some test cases run individually and a comment of the uint64_t ret value in C:
let ret=0; // 0x0
let ret=1; // 0x3ff0000000000000
let ret=-1; // 0x3ff0000000000000
let ret=2; // 0x4000000000000000
let ret=3; // 0x4008000000000000
let ret=4; // 0x4010000000000000
let ret='1'; // 0xfff6000000003101
let ret=2>>1; // 0x3ff0000000000000
let ret=2>>>1; // 0x3ff0000000000000

We tried using mjs_own(mjs, &ret); with no change in effect.

IDE: KEIL
Compiler: ARMCC
MCU: Nordic nrf52832

jnz86, 2018-01-27:
Extra info:

Board: Nordic nrf52832-DK
Compiler: ARMCC 4.5 and 5.2
Nordic Includes: SDKv14.2 and DFP 8.15.2

I can check/post the disassembly if needed.

dimonomid, 2018-01-29:
`mjs_val_t` is a NaN-packed internal mjs value, its value is not directly human-readable. Use [`mjs_get_double()`](https://github.com/cesanta/mjs/blob/master/mjs.h#L852) to get a number value from it, or ['mjs_get_string()`](https://github.com/cesanta/mjs/blob/master/mjs.h#L972) to get a string, etc.

closed 2018-01-29 by dimonomid