cesanta/mjs #82: stdbool.h missing from some platforms (nrf52)

jnz86, 2018-01-26:
Building mjs using the nrf52 platform. The "guesser" at line #32 has no support for nrf52, so using the define of CS_PLATFORM=CS_P_NRF52. 

stdbool.h is not brought in near line 1161 with the rest of the other includes for the nrf52 platform. Must be manually added every release.

stdbool.h is included with the ESP32, Ti CC, and other platforms in their respective sections.

IDE: KEIL
Compiler: ARMCC
MCU: Nordic nrf52832

dimonomid, 2018-01-29:
Since you have ARMCC compiler ready, could you please create a pull request which would fix the problem for you, so that we can integrate it? Thanks

closed 2018-01-29 by rojer

dimonomid, 2018-01-29:
Thanks!

We can't merge PRs directly because this is an autogenerated repo, but I integrated it manually into our internal repo, it'll be published here soon.