cesanta/mjs #83: function(arguments) limits?

Spiritdude, 2018-01-27:
I noticed that whenever I compose something like:
```
someFunction("here and now" + String.format("%d",1+2) + something_else + ... );
```
the same with 
```
RPC.call(..., function(resp, err, err_msg, ud) {
  // - 30 lines of code
}, usercode);
```
when I `load` the code on demand, I get a core dump like
```
[Jan 27 16:19:27.417] Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
[Jan 27 16:19:27.423] Debug exception reason: Stack canary watchpoint triggered (mgos)
[Jan 27 16:19:27.429] Register dump:
....
```
and I end up changing my code, and replace the arguments with variable(s) instead of code:
```
let s = "here and now" + ....;
someFunction(s);

let f = function(resp,err, ... ) {
 ....
};
RPC.call(....,f,null);
```
and things work fine.

As if there a limit how much complexity and length in the argument list in a function call is allowed.

cpq, 2018-01-29:
This could be a stack size issue, not a complexity issue.
Parsing JS code needs stack space.
I assume, you're working on ESP8266?

Spiritdude, 2018-01-29:
I work with ESP8266 (Wemos D1 Mini) and ESP32 (LOLIN32 Lite), I can't recall exactly if I had no problems with ESP32 since I test my code on both; and resolved it for ESP8266 and later tested on ESP32 as well. You saw the other issue https://github.com/cesanta/mjs/issues/80 on JSON.parse() I posted, there definitely both have problems (ESP8266 more, ESP32 less but also has problems), not sure if this related; it seemed to me function calls in arguments of function calls I avoid when large(r) data chunk is handled, but assign it first to a variable and hand over that variable.