cesanta/mjs #88: Custom function implementation?

NaxAlpha, 2018-03-05:
Hi,
I would like to manually implement JavaScript function within C without using FFI
What I need is something like this:
``` C
void add(struct mjs* inst) {
	int a = mjs_get_int(inst, mjs_arg(inst, 0));
	int b = mjs_get_int(inst, mjs_arg(inst, 0));
	int c = a + b;
	mjs_return(inst, mjs_mk_number(inst, c));
}
```
And I want to use it directly in JavaScript (without ffi) like `add(1,2)`
Thanks

cpq, 2018-03-05:
So ?

NaxAlpha, 2018-03-06:
Thanks for the help!
The problem was, there is no documentation on how to implement function manually without using ffi. I was able to follow the original code and implemented successfully:
``` C
void add(mjs* m) {
	int x = mjs_get_int(m, mjs_arg(m, 0));
	int y = mjs_get_int(m, mjs_arg(m, 1));
	int z = x + y;
	mjs_val_t ret = mjs_mk_number(m, z);
	mjs_return(m, ret);
}
int main() {
	struct mjs* mjs = mjs_create();
	auto ff = mjs_mk_foreign_func(mjs, (mjs_func_ptr_t)add);
	mjs_set(mjs, mjs_get_global(mjs), "add", ~0, ff);
	mjs_val_t ret;
	mjs_exec(mjs, "add(1, 2)", &ret);
	int num = mjs_get_int(mjs, ret);
	mjs_destroy(mjs);
	printf("Result:%d", num);
	return 0;
}
```

closed 2018-03-06 by NaxAlpha