cesanta/mjs #96: Need a way to get/set userdata

jockm, 2018-06-28:
There needs to be a way to associate a pointer to user data to a specific instance of mJS, similar to what duktape has with their udata pointer.

This way a mJS instance could contain a pointer to additional data, the object that istanciated it, etc.

In our case we may have multiple mJS instances operating at once.  Therefor the native functions need to know what C++ object is associated with the specific call.  While we could pass that into the VM and then have ffi pass that back to us, it would simply be cleaner and safer to have associated userdata.

Right now we are storing this information in the VM under an obfuscated name, but this is not a good long term strategy

cpq, 2018-06-29:
Could that be done by keeping a map mjs_pointer -> cpp_object ?

jockm, 2018-06-29:
I don't like the idea of having this stored in any way that is visible to the script inside the VM.  It opens a potential security hole.

I really want an additional field added to the `struct mjs`.  This way I can associate user data with the vm, but not have it visible in any way.  

We aren't using FFI for our native functions, so we can get the `struct mjs *` passed to us.  I never want the scripts even *aware* that there is something that identifies the object instance in them.  Remove the vector before it exists if you will