cesanta/mjs #101: please add .toString()

jockm, 2018-06-30:
Short of using `JSON.stringify(...)` there is no apparent way to convert a number to a string.  This is a long way round just to get a glass of water.

While there is't a Sting class per, se; mJS supports other methods on strings.  String conversion is vital for generating messages, debugging, etc.

tripflex, 2018-08-09:
I agree this would be awesome, I have TONS of `JSON.stringify` calls in my code would be much cleaner code base AND easier for JS devs to dive into mJS by having this available 👍

cpq, 2018-08-10:
adding parseInt makes sense.
the number -> string conversion, what's the suggestion ?

jockm, 2018-08-10:
> the number -> string conversion, what's the suggestion ?

To add back the `.toString()` method.  For strings it would return itself, for numbers it would return the stringified number, for everything else it could return `Object` or `Unsupported` or the like.

Yes this would break it being a strict subset, but right now you write nonconformant code to get around this, or you write your own number parser in Javascript which just moves the code size to the user.

cpq, 2018-08-14:
What is wrong with `JSON.stringify()` though ?
is it too long ? how about making a `str()` shortcut ?

jockm, 2018-08-14:
Look I value the fact that you want mJS to be both a strict subset and as small as it can be; but you can be too small.  IMHO mJS passed too small some time ago.  `.toString()` is the logical, language based way to do this.  It is what someone who writes JavaScript aches to do.

Using `JSON.stringify()` is a hack plain and simple, and proposing it as the official way to convert something to be a string comes off as being somewhere between lazy and and defending artificial smallness.

mJS's subset is so small that it is effectively a different language at this point.  If you want to be a ES5-like language then fine, double down and go that route.  I am sure there are other things you could remove to make it even smaller.

However if you want to be ES5, that means not just conforming to some arbitrary syntax, but to the spirit and idioms of the language.  We give up so much just to use mJS, possibly too much.  Give us this