cesanta/mjs #118: Can i make mjs work on CC2538 TI Zigbee microcontroller?

dejan994, 2018-09-03:
Hello,

Do you know if i can make mjs work on CC2538 with 512KB flash and 32KB RAM? I tried it on bare metal system without OS and it works flawlessly, but the problem is when i try to integrate it into the OSAL system (small operating system for CC2538). In debugging section it just drops into exit and won't go through code. It stops on this line:

cur->head.link = a->free; which is located in gc_new_block function.

I wonder if this is some problem with my integration, or is it just lack of RAM.

Best regards

cpq, 2018-09-03:
Maybe the problem is with the concurrent access. Are you using mjs from different RTOS tasks or something? There is no internal locking, so make sure to serialize mjs access.

dejan994, 2018-09-03:
I'm calling the function from the events section in OSAL, which uses tasks. What do you mean by serializing mjs access? I tried putting it into beginning of the main, and the same thing happens.

cpq, 2018-09-03:
Mutex-protect all code sections that use mjs.

dejan994, 2018-09-03:
I put this code in my main:

int main( void )
{
   
    struct mjs *mjs = mjs_create();
    mjs_set_ffi_resolver(mjs, js_led);
    mjs_exec_file(mjs, "led.js", NULL);
    return 0;

The problem is, when i set breakpoints and debug it, it goes into first row, but won't go into second function mjs_set_ffi_resolver(mjs, js_led);. Is that really the problem here , if it goes into the first function? 
Do i need mutex even for first command in main?

cpq, 2018-09-03:
I think the problem is elsewhere - like, malloc not initialised or something. If you can get a stack backtrace where it hangs, that'd be great.

cpq, 2018-09-03:
Could you dump the whole backtrace please?
Apparently it is in the busy loop.

dejan994, 2018-09-04:
I'm sorry for answering so late, i wasn't at work so i couldn't post stack trace. 
It goes like this (top as the last one):

[__exit + 0xb]
gc_new_block
gc_arena_init
mjs_create
main
[_call_main + 0x9]

cpq, 2018-09-10:
Yes, apparently this block is in the busy loop:

https://github.com/cesanta/mjs/blob/2.5.1/mjs/src/mjs_gc.c#L97-L102

I assume the problem is with the malloc implementation used in the RTOS.
Could you check the free RAM when it hangs, btw?