cesanta/mjs #125: Is mJS being actively developed?

jockm, 2019-02-13:
It doesn't appear that a bug has been closed in 9 months, no new issue has been responded to in months.

But mJS doesn't feel done, at least not to me.  There needs to be an official way to omit unneeded/dangerous builtins short of modifying the code.  What about a series of #defines that allow me to add/remove features like `load()`, and `die()`, and the like?

The #defines for memory management should take an additional parameter so that you can use a memory pool based allocator more easily.

The JSON library should use the same #defines for memory management as mJS.

mJS shouldn't depend on standard io.  Being so memory lean it can be used in embedded apps that don't have a supporting (RT)OS, and thus may not have <stdio> like file functions.  Imagine trying to integrate Elm-Chan's  [Petite FAT Filesystem](http://elm-chan.org/fsw/ff/00index_p.html) with mJS.

All of this is before you get to that in an ideal world there should be a way to integrate more of [v7](https://github.com/cesanta/v7)'s language features in a scalable way. So that you can go from mJS extreme subset, all the way up to something much more familiar if you have the resources for it.  See how [Duktape](https://duktape.org/) does this, for example.  Alternatively give us an official way to do this ourselves, and document the internal API.

Right now the subset of JavaScript that is supported is minimal that it doesn't really look like the language most people use.  But it would be nice to use the `Math` object instead of ffi — to name one example — so that simple JavaScript libraries can be ported over more easily. 

Every single issue I just named is something my team personally dealt with, and that ultimately prevented us from paying for a commercial license for mJS.  I also don't think any of them are unreasonable enhancements.  

However if mJS isn't being actively developed, then please release mJS under the LGPL.  That way there will still be an incentive for companies to purchase a closed license, and it will be more desirable for folk like myself to enhance the code

jockm, 2019-03-25:
It has now been well over a month and there has been no response.  I can only conclude that mJS has been effectively abandoned.  That is a shame...

cpq, 2019-03-26:
mJS is used as a scripting language in https://github.com/cesanta/mongoose-os and does its job well.

All the features you've mentioned, @jockm , are reasonable, but do not justify to assign a developer on them, since we're a business.

Thus, either sign a CLA and send PRs, or move to some other solution.

closed 2019-03-26 by cpq

jockm, 2019-03-26:
Well it is nice to see an official response.  It is you call, but it is just a shame.