cesanta/mjs #134: mjs_destroy crashes ESP32 when using Timer

T94T, 2019-04-23:
Hi,

I'm calling some JavaScript dynamically:
```
mjs_exec(mjs, code, NULL);
```
code is for example:
```load("api_timer.js");
Timer.set(1000, true, function() {
  print(".");
}, null);
```
I need to dispose the MJS instance from time to time and I'm calling:
```
mjs_destroy(mjs);
```
This then crashes the ESP32 and I get:

[error.txt](https://github.com/cesanta/mjs/files/3106341/error.txt)

It totally works fine when using code without Timers though!

cpq, 2019-05-03:
try `let t = Timer.set(/* ... */);`, before disposing, `Timer.del(t);`

TOLDOTECHNIK, 2019-05-04:
Thank you! This code works
```
load("api_timer.js");
let t = Timer.set(1000, true, function() {
  print("tick");
}, null);

Timer.set(5000, false, function(){
  Timer.del(t);
}, t);
```
After 5 seconds I can run `mjs_destroy(mjs)` without crashing my ESP.

But I need to call `Timer.del(t)` from the outside of the current MJS instance (in C). Is that possible?

T94T, 2019-05-04:
Thank you! This code works
```
load("api_timer.js");
let t = Timer.set(1000, true, function() {
  print("tick");
}, null);

Timer.set(5000, false, function(){
  Timer.del(t);
}, t);
```
After 5 seconds I can run `mjs_destroy(mjs)` without crashing my ESP.

But I need to call `Timer.del(t)` from the outside of the current MJS instance (in C). Is that possible?