cesanta/mjs #175: Heap-based Buffer Overflow Vulnerability

C0NSTANTINE110, 2021-06-30:
### Built:
Jun 30 2021

### Details:
heap-based buffer overflow **mjs.c:7617** in **mjs_set_errorf**

### Command:
./mjs -f Heap_Buffer_Overflow.js

### Result:
==2419050==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x604000000178 at pc 0x55555557f3ed bp 0x7fffffffcf40 sp 0x7fffffffcf30
READ of size 8 at 0x604000000178 thread T0
    #0 0x55555557f3ec in mjs_set_errorf /home/constantine/mjs/mjs.c:7617
    #1 0x555555598395 in parse_literal /home/constantine/mjs/mjs.c:12166
    #2 0x55555559861b in parse_call_dot_mem /home/constantine/mjs/mjs.c:12175
    #3 0x5555555990d3 in parse_postfix /home/constantine/mjs/mjs.c:12209
    #4 0x55555559932c in parse_unary /home/constantine/mjs/mjs.c:12228
    #5 0x5555555995d1 in parse_mul_div_rem /home/constantine/mjs/mjs.c:12241
    #6 0x555555599ba8 in parse_plus_minus /home/constantine/mjs/mjs.c:12246
    #7 0x55555559a1c1 in parse_shifts /home/constantine/mjs/mjs.c:12251
    #8 0x55555559a648 in parse_comparison /home/constantine/mjs/mjs.c:12255
    #9 0x55555559a9bb in parse_equality /home/constantine/mjs/mjs.c:12259
    #10 0x55555559ae46 in parse_bitwise_and /home/constantine/mjs/mjs.c:12264
    #11 0x55555559b3ec in parse_bitwise_xor /home/constantine/mjs/mjs.c:12269
    #12 0x55555559b992 in parse_bitwise_or /home/constantine/mjs/mjs.c:12274
    #13 0x55555559bf38 in parse_logical_and /home/constantine/mjs/mjs.c:12279
    #14 0x55555559c4de in parse_logical_or /home/constantine/mjs/mjs.c:12284
    #15 0x7fffffffdc0f  ([stack]+0x1fc0f)

Address 0x604000000178 is a wild pointer.
SUMMARY: AddressSanitizer: heap-buffer-overflow /home/constantine/mjs/mjs.c:7617 in mjs_set_errorf
Shadow bytes around the buggy address:
  0x0c087fff7fd0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c087fff7fe0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c087fff7ff0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c087fff8000: fa fa 00 00 00 00 00 00 fa fa 00 00 00 00 00 00
  0x0c087fff8010: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
=>0x0c087fff8020: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa[fa]
  0x0c087fff8030: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c087fff8040: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c087fff8050: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c087fff8060: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c087fff8070: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==2419050==ABORTING

### PoC:
[Heap_Buffer_Overflow.js.tar.gz](https://github.com/cesanta/mjs/files/6738729/Heap_Buffer_Overflow.js.tar.gz)

rojer, 2021-06-30:
so, the problem seems to be in allowing unary `--` on `print`.
minimum repro is:
```
--print;
print(1);
```
this segfaults in mjs_builtin call, presumably `print` having been corrupted by `--`.
@cpq maybe you can help with adding validation so `print` doesn't actually get tampered with?

not sure what the correct behavior should be here, fwiw if you `--` a function in V8 it disappears from the namespace.

X3eRo0, 2021-07-27:
How do you fuzz js code? how do you mutate and generate valid js code?

C0NSTANTINE110, 2021-07-27:
> How do you fuzz js code? how do you mutate and generate valid js code?

It's compiled by afl-gcc and processed by afl. Generating the the valid js code is not the point, the flaw causes a denial-of-service (and potentially code execution) due to a heap buffer overflow which can be triggered via a crafted js file.

X3eRo0, 2021-07-27:
You are telling me that the crashing poc .js file is generated by AFL? I thought it's some grammer fuzzing or something like that